bouffalolab/bouffalo_sdk #2: which riscV mcu ?

netmonk, 2021-06-08:
why did you work so hard to remove gd32f103V reference in the documentation ? 
It's pretty clear the riscV mcu is this one on your SOC.

robertlipe, 2021-06-09:
Bouffalo makes SOCs that are similar to the GD32V. They don't use the same peripheral sets. They don't even license the RISC-V cores from the same vendor (Nuclei vs. SiFive). (I've programmed both.) Lots of the middleware (e.g. FatFS, LGVL, etc.) will run on any part of this class, including ARM cores, so that code will look familiar.

I don't know if you think that Bouffalo is rebadging or even cloning the GD32V parts, but they're just not.  The libfat they're providing doesn't include code for the GD32V, but that doesn't mean they removed it - it came from the same beginnings and both vendors acknowledge that. The ChaN code is a good implementation that shows up on mcus everywhere from 8051 up.

It's clear that the Boufallo parts are RISC-V, just like the GD32V is.  There is no "aha!" Scooby-Doo reveal there. That's clearly stated by both of them. It's an open source IP and there are dozens or maybe even hundreds of implementations of RISC-V at this point.

What are you talking about?

If we take the title literally and try to answer that, we can answer that from the first sentence in README.md even without speaking any Chinese. This repository contains code for the BL602, BL604, BL702, BL704, and BL706 by Bouffalo Labs - the "BL" in those numbers.  Bouffalo isn't a well known company in the Western world - hell, I don't know if they're well known within China - but they're an actual company that makes actual chips. https://www.cnx-software.com/2020/10/24/bl602-bl604-risc-v-wifi-bluetooth-5-0-le-soc-will-sell-at-esp8266-price-point/ BL60x has been shipping for a while and is in many products that would have probably otherwise had ESP8266-class devices while BL70x is pretty new.

Be more respectful if you're asking an actual question. Have actual evidence if you're making an accusation. Your post is like asking StarFive or Kendryte why they removed GD32V references.

netmonk, 2021-06-09:
Thank you Very much Robert for your answer. I was a little rude i admit. And your explanation are clearly explaining differences between buffalo and gdf32vf103 platform.

closed 2021-06-20 by profound1987