bouffalolab/bouffalo_sdk #6: BL602 : unknown type names : Efuse_Ldo11VoutSelTrim_Info_Type & Efuse_TxPower_Info_Type

JF002, 2021-07-02:
Hello!

I'm trying to use this SDK to build an app for the BL602 (Pine64 board). I tried to build the helloworld app as follows:

```
make BOARD=bl602_iot CHIP=bl602 APP=helloworld
```

But I get build errors related to `Efuse_Ldo11VoutSelTrim_Info_Type` and `Efuse_TxPower_Info_Type` that are unknown. Here's the full build log:

```
$ make BOARD=bl602_iot CHIP=bl602 APP=helloworld
cmake -E make_directory build
cmake -E copy project.build build
make -C ./build -f ./project.build
make[1]: Entering directory '/home/jf/git/bl_mcu_sdk/build'
cmake -DCHIP=bl602 -DCPU_ID=m0 -DBOARD=bl602_iot -DSUPPORT_FLOAT=n -DSUPPORT_SHELL=n -DSUPPORT_FREERTOS=n -DSUPPORT_CRYPTO=n -DSUPPORT_LVGL=n -DSUPPORT_BLE=n -DSUPPORT_XZ=n -DSUPPORT_LWIP=n -DSUPPORT_TFLITE=n -DAPP_DIR=examples -DAPP=helloworld ..
-- TOOLCHAIN_PATH:/home/jf/bl602/riscv64-elf-x86_64-20210120/bin/
-- TOOLCHAIN_PATH:/home/jf/bl602/riscv64-elf-x86_64-20210120/bin/
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/jf/bl602/riscv64-elf-x86_64-20210120/bin/riscv64-unknown-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/jf/bl602/riscv64-elf-x86_64-20210120/bin/riscv64-unknown-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/jf/bl602/riscv64-elf-x86_64-20210120/bin/riscv64-unknown-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- [run app:helloworld], path:/home/jf/git/bl_mcu_sdk/examples/hellowd/helloworld
-- [register library component: common], path:/home/jf/git/bl_mcu_sdk/common
-- [register library component: fatfs], path:/home/jf/git/bl_mcu_sdk/components/fatfs
-- [register library component: usb_stack], path:/home/jf/git/bl_mcu_sdk/components/usb_stack
-- [register library component: bl602_driver], path:/home/jf/git/bl_mcu_sdk/drivers/bl602_driver
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CPU_ID

-- Build files have been written to: /home/jf/git/bl_mcu_sdk/build
make -j4
make[2]: Entering directory '/home/jf/git/bl_mcu_sdk/build'
make[3]: Entering directory '/home/jf/git/bl_mcu_sdk/build'
make[4]: Entering directory '/home/jf/git/bl_mcu_sdk/build'
make[4]: Entering directory '/home/jf/git/bl_mcu_sdk/build'
make[4]: Entering directory '/home/jf/git/bl_mcu_sdk/build'
make[4]: Leaving directory '/home/jf/git/bl_mcu_sdk/build'
make[4]: Leaving directory '/home/jf/git/bl_mcu_sdk/build'
make[4]: Leaving directory '/home/jf/git/bl_mcu_sdk/build'
make[4]: Entering directory '/home/jf/git/bl_mcu_sdk/build'
make[4]: Entering directory '/home/jf/git/bl_mcu_sdk/build'
make[4]: Entering directory '/home/jf/git/bl_mcu_sdk/build'
[  1%] Building C object libraries/usb_stack/CMakeFiles/usb_stack.dir/class/audio/usbd_audio.c.obj
[  3%] Building C object libraries/fatfs/CMakeFiles/fatfs.dir/diskio.c.obj
[  4%] Building C object libraries/common/CMakeFiles/common.dir/device/drv_device.c.obj
[  6%] Building C object libraries/common/CMakeFiles/common.dir/memheap/drv_mmheap.c.obj
[  7%] Building C object libraries/usb_stack/CMakeFiles/usb_stack.dir/class/cdc/usbd_cdc.c.obj
[  9%] Building C object libraries/fatfs/CMakeFiles/fatfs.dir/ff.c.obj
[ 11%] Building C object libraries/common/CMakeFiles/common.dir/misc/misc.c.obj
[ 12%] Building C object libraries/usb_stack/CMakeFiles/usb_stack.dir/class/hid/usbd_hid.c.obj
[ 14%] Building C object libraries/usb_stack/CMakeFiles/usb_stack.dir/class/msc/usbd_msc.c.obj
[ 15%] Building C object libraries/common/CMakeFiles/common.dir/partition/partition.c.obj
[ 17%] Building C object libraries/usb_stack/CMakeFiles/usb_stack.dir/class/video/usbd_video.c.obj
[ 20%] Building C object libraries/usb_stack/CMakeFiles/usb_stack.dir/core/usbd_core.c.obj
[... 77 lines cut ...]
         if (trim->parity == EF_Ctrl_Get_Trim_Parity(trim->txpower, 5)) {
                                                         ^~
/home/jf/git/bl_mcu_sdk/drivers/bl602_driver/std_drv/src/bl602_ef_ctrl.c:423:21: error: request for member 'txpower' in something not a structure or union
             if (trim->txpower >= 16) {
                     ^~
/home/jf/git/bl_mcu_sdk/drivers/bl602_driver/std_drv/src/bl602_ef_ctrl.c:424:32: error: request for member 'txpower' in something not a structure or union
                 *TxPower = trim->txpower - 32;
                                ^~
/home/jf/git/bl_mcu_sdk/drivers/bl602_driver/std_drv/src/bl602_ef_ctrl.c:426:32: error: request for member 'txpower' in something not a structure or union
                 *TxPower = trim->txpower;
                                ^~
[ 57%] Building C object libraries/bl602_driver/CMakeFiles/bl602_driver.dir/std_drv/src/bl602_glb.c.obj
make[4]: *** [libraries/bl602_driver/CMakeFiles/bl602_driver.dir/build.make:285: libraries/bl602_driver/CMakeFiles/bl602_driver.dir/std_drv/src/bl602_ef_ctrl.c.obj] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: Leaving directory '/home/jf/git/bl_mcu_sdk/build'
make[3]: *** [CMakeFiles/Makefile2:274: libraries/bl602_driver/CMakeFiles/bl602_driver.dir/all] Error 2
make[3]: Leaving directory '/home/jf/git/bl_mcu_sdk/build'
make[2]: *** [Makefile:91: all] Error 2
make[2]: Leaving directory '/home/jf/git/bl_mcu_sdk/build'
make[1]: *** [project.build:18: __build] Error 2
make[1]: Leaving directory '/home/jf/git/bl_mcu_sdk/build'
make: *** [Makefile:44: build] Error 2
```

I'm using the current master branch (tag v1.2.3 from 1 July).

shchen-Lab, 2021-07-03:
Hello,
Thanks for your feedback! This‘s a driver issue, we will fix this issue soon.

JF002, 2021-07-03:
FYI, I've just added those declaration (coming from https://github.com/bouffalolab/bl_iot_sdk) in drivers/bl602_driver/std_drv/inc/bl602_ef_ctrl.h, and it seems to build and run fine : 

```
/**
 *  @brief Efuse Ldo11 Vout Sel Trim definition
 */
typedef struct {
    uint32_t sel_value                              :  4;    /*!< value trim */
    uint32_t parity                                  :  1;    /*!< Parity of capcode */
    uint32_t en                                      :  1;    /*!< Enable status */
    uint32_t rsvd                                    : 26;    /*!< Reserved */
}Efuse_Ldo11VoutSelTrim_Info_Type;

/**
 *  @brief Efuse Tx Power definition
 */
typedef struct {
    uint32_t txpower                                 :  5;    /*!< txpower value  */
    uint32_t parity                                  :  1;    /*!< Parity of capcode */
    uint32_t en                                      :  1;    /*!< Enable status */
    uint32_t rsvd                                    : 25;    /*!< Reserved */
}Efuse_TxPower_Info_Type;
```

closed 2021-07-05 by profound1987