bouffalolab/bouffalo_sdk #9: [Feature Request]: Please Bl

cooldome, 2021-07-07:
Please add  "sipeed RV plus Debugger BL702" board support as mcu target.
It would allow for the favorable setup, use two sipeed BL702 boards: one as mcu and one as debugger.

Thank you

sakumisu, 2021-07-09:
bl702 default JTAG pin are list：TMS(GPIO0)、TDI(GPIO1)、TCK(GPIO2)、TDO(GPIO9),so if you wants to connect two boars as JTAG debugger for one another,you should connect one TMS、TDI、TCK with another TDI、TDO、TMS，then there is TDO with TCK left, TDO is GPIO9,and is led0 on rv debugger ,its location is align with TMS pin on board,so you need to solder a wire to the resistor right on the right side of the led and link this pin to JTAG TCK. Finally ,you can debug with two boards.

strongwong, 2021-07-14:
https://bbs.bouffalolab.com/d/61-rvddebuggeropenocdbl706/7
There are pictures in this post to illustrate, you can refer to it.

closed 2021-07-16 by profound1987