bouffalolab/bouffalo_sdk #22: BL70X IoT DVK-MB-V1.1 cannot be flashed

gamelaster, 2022-02-25:
I have BL70X IoT DVK-MB-V1.1, which have BL706 with no internal FLASH or RAM, everything is external.
It have Winbond 25Q32JVS1Q and 1604M-3SQR. 

Unfortunately, I can't get this flashed with bflb_mcu_tool or Bouffalo Lab Dev Cube, it all ends on `Flash load shake hand`.
I tried to change `flash_id = ef8016` (which is flash config for `W25Q32JW_ef8016.conf`), although, the eflash_loader still can't load the SPI flash. Any ideas?

Thanks

gamelaster, 2022-03-01:
Update: I get it working. The problem is USB with combination of Windows. For some reason, after `eflash_loader` is executed, the USB CDC COM port is locked out and flasher can't communicate with it. It works OK on Linux.

sakumisu, 2022-03-01:
This problem has not been encountered. It may be that your flash type is not support in devcube, and the eflash_loader ini and cfg are automatically modified, please do not manually modify.If you use usb to download fail,you can use serial port to have a try.Thanks.

sakumisu, 2022-03-01:
If you can, you can send an email to this address : jzlv@bouffalolab.com.Please provide flash download log, it is under devcube directory named log.Thanks.

gamelaster, 2022-03-01:
Hi @sakumisu ,

I sent you an email with the log. The UART Flash works OK, and USB Flash works OK on Linux.
I talked with Lup Yuen, and he told me that flashing works good on Windows 10.
But I have Windows 11, so my suspicion is that Windows 11 have some issues with USB CDC driver.
I will test it on Windows 10 and Windows 7 soon.
Thanks

gamelaster, 2022-03-01:
Hi @sakumisu ,
I confirmed it, flashing doesn't work on Windows 11, but works on Windows 10.
On Windows 7, there is driver missing.

gamelaster, 2022-06-02:
Together with @Ralim we analyzed this issue (we we're able to reproduce it also on Linux), and able to fix it, so flashing works reliably both on Linux and Windows 10 / 11. Looks like `img_load_create_predata_before_run_img` which was added in recent version of flasher fixes the issue.

Anyway, I will create separate repository for Windows 7 driver.

Thanks

closed 2022-06-02 by gamelaster

turboJo, 2022-06-02:
这是来自QQ邮箱的假期自动回复邮件。您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。