bouffalolab/bouffalo_sdk #26: BLE example build seems to be broken.

Jatsekku, 2022-03-30:
I tried to compile ble_peripheral example and get:

```
[ 97%] Linking C executable ../../../out/examples/ble/ble_peripheral/ble_peripheral_bl702.elf
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(hci_onchip.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(hci_onchip.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../../components/ble/blecontroller/lib/libblecontroller.a(hci_onchip.o)
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(ke_msg.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(ke_msg.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../../components/ble/blecontroller/lib/libblecontroller.a(ke_msg.o)
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(ke_task.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(ke_task.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../../components/ble/blecontroller/lib/libblecontroller.a(ke_task.o)
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(arch_main.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(arch_main.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../../components/ble/blecontroller/lib/libblecontroller.a(arch_main.o)
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(em_buf.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(em_buf.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../../components/ble/blecontroller/lib/libblecontroller.a(em_buf.o)
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(lld.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(lld.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../../components/ble/blecontroller/lib/libblecontroller.a(lld.o)
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(llm.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(llm.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../../components/ble/blecontroller/lib/libblecontroller.a(llm.o)
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(rwble.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(rwble.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../../components/ble/blecontroller/lib/libblecontroller.a(rwble.o)
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(ea.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(ea.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../../components/ble/blecontroller/lib/libblecontroller.a(ea.o)
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(rwip.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(rwip.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../../components/ble/blecontroller/lib/libblecontroller.a(rwip.o)
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(ke_event.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(ke_event.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../../components/ble/blecontroller/lib/libblecontroller.a(ke_event.o)
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(ke_mem.o): conflicting priv spec version (major/minor/revision).
/usr/bin/riscv64-elf-20210120/bin/../lib/gcc/riscv64-unknown-elf/8.4.0/../../../../riscv64-unknown-elf/bin/ld: error: ../../../components/ble/blecontroller/lib/libblecontroller.a(ke_mem.o): conflicting priv spec version (major/minor/revision).
[... cut ...]
make[3]: *** [samples/ble_peripheral/CMakeFiles/ble_peripheral_bl702.elf.dir/build.make:153: ../out/examples/ble/ble_peripheral/ble_peripheral_bl702.elf] Error 1
make[3]: Leaving directory '/home/ubuntu/bl_mcu_sdk/build'
make[2]: *** [CMakeFiles/Makefile2:202: samples/ble_peripheral/CMakeFiles/ble_peripheral_bl702.elf.dir/all] Error 2
make[2]: Leaving directory '/home/ubuntu/bl_mcu_sdk/build'
make[1]: *** [Makefile:84: all] Error 2
make[1]: Leaving directory '/home/ubuntu/bl_mcu_sdk/build'
make: *** [Makefile:51: build] Error 2
```
I'm starting loosing my mind working with this bl702 mcu, it's seems to be too soon for real development...

sakumisu, 2022-04-04:
Update your toolchain to 10.2. url ：https://gitee.com/bouffalolab/toolchain_gcc_t-head_linux

Jatsekku, 2022-04-09:
Additional info about problem at internal BL's forum: https://bbs.bouffalolab.com/d/129-bl-mcu-sdk-ble-apps-build-problem
and here (under point 24): https://bbs.bouffalolab.com/d/66-bl-mcu-sdk/8 
After update to SiFive, BLE example works like a charm.

closed 2022-04-09 by Jatsekku