bouffalolab/bouffalo_sdk #32: BL808 JTAG documentation

btashton, 2022-10-30:
There does not seem to be much in the way of documentation for using JTAG with the BL808 chip.  I have been trying to use a RV Debugger Plus with a Ox64 board but am not able to examine any of the cores.

gamelaster, 2022-10-31:
Hi @btashton is very likely you received prototype ox64 board, so the pinout of JTAG is not correct.

The correct pins for JTAG are 6,7,12,13.

12, 13 should be on JTAG header, 6, 7 on GPIO headers.

btashton, 2022-10-31:
There are three different ones (the LP is a two wire debug). You can actually see the mappings in the datasheet ( it is not reference manual).

btashton, 2022-11-02:
I am going to close this now that documentation for the CKLink has been provided.  You can find my notes here:
https://gist.github.com/btashton/6af120ff16d6beeccbbde74e6733535c

closed 2022-11-02 by btashton

davidcorrigan714, 2022-12-10:
Either this needs to be re-opened or the title changed to reflect something about debugging. We very much still need JTAG documentation and something working with OpenOCD and not some random bespoke debugger for this chip to be really broadly useful.

btashton, 2022-12-10:
I would recommend this
 https://btashton.github.io/bl808-notes/jtag.html
And
https://btashton.github.io/bl808-notes/devboards/ox64.html

However I stopped using the Ox64 because my early  board seemed to have issues with with flash and sipeed got me a functional M1s. I will come back to the Ox64 when a new one arrives.

gamelaster, 2022-12-11:
@btashton You have 16Mb flash, which is not faulty. Just you need to set proper Flash configuration in BL DevCube to make it work.

gamelaster, 2022-12-11:
@btashton if you have issues with flashing of Ox64 with 16Mb flash, it is because inproper configuration of SPI Flash in BLDevCube. I can share the configuration if needed (I'm AFK currently)

btashton, 2022-12-11:
> @btashton if you have issues with flashing of Ox64 with 16Mb flash, it is because inproper configuration of SPI Flash in BLDevCube. I can share the configuration if needed (I'm AFK currently) 

Sure. I can give it a try, but seemed like something was not right with the timing parameters for the smaller flash.

gamelaster, 2022-12-11:
@btashton 
Firstly, replace flashcfg_list.csv in utils/flash/bl808 folder. ([flashcfg_list.csv](https://github.com/bouffalolab/bl_mcu_sdk/files/10203170/flashcfg_list.csv))
Then, open file chips/bl808/eflash_loader/eflash_loader_cfg.ini, and find flash_id entry. Ensure, that it is configured as following:
[image]

After that, it should work fine :)

- gamiee

btashton, 2022-12-11:
I will try sometime this week and update my notes. Thanks!

gamelaster, 2022-12-11:
No problem :)

gamelaster, 2022-12-17:
@btashton sorry, I written you wrong docs, I updated ti with my factory test firmware. It outputs text into UART + toggling all GPIO pins.

davidcorrigan714, 2022-12-26:
The programmer I ordered got in a few days ago and I just got that factory_test bin loaded and working. The GPIOs are toggling and the UART is seeing a stream of Ping! and Pong! .

NilsBe, 2023-02-20:
> @btashton Firstly, replace flashcfg_list.csv in utils/flash/bl808 folder. ([flashcfg_list.csv](https://github.com/bouffalolab/bl_mcu_sdk/files/10203170/flashcfg_list.csv)) Then, open file chips/bl808/eflash_loader/eflash_loader_cfg.ini, and find flash_id entry. Ensure, that it is configured as following: [image] After, open Bouffalo Lab Dev Cube, select BL808, and navigate to MCU tab. Set M0 to group0, Image Addr 0x58000000 , and specify path to factory_test_bl808_m0.bin. Then Create & Download.
> 
> [factory_test_bl808_m0.bin.tar.gz](https://github.com/bouffalolab/bl_mcu_sdk/files/10251417/factory_test_bl808_m0.bin.tar.gz) (don't forget to extract it)
> 
> After that, it should work fine :)
> 
> * gamiee

Sorry, doesn't work in my case. 
Also, I would like to ask where I can find the source for factory_test.bl808_m0.bin

DavidVentura, 2023-02-25:
I've also applied the recommendations from @gamelaster on the `bflb_mcu_tool`, replaced the CSV with the linked one and changed the values in FLASH_CFG so that they look like

```
[FLASH_CFG]
#flash_id = ef4016 # <- was the original value
flash_id = ef6015
#bit 7-4 flash_clock_type: 0:120M wifipll, 1:xtal, 2:128M cpupll, 3:80M wifipll, 4:bclk, 5:96M wifipll
#bit 3-0 flash_clock_div
#flash_clock_cfg = 0x41 # <- was the original value
flash_clock_cfg = 0x43

# everything below here didn't change
#0:0.5T delay, 1:1T delay, 2:1.5T delay, 3:2T delay
flash_clock_delay = 0
#0:NIO, 1:DO, 2:QO, 3:DIO, 4:QIO
flash_io_mode = 1
#flash_pin value:
#bit 7 flash pin set from efuse flash cfg
#bit 6-0 flash pin cfg: 0: internal swap io0 with io3, 2: internal no swap io0 with io3, 4: external gpi34-39,
flash_pin = 0x80
#empty for auto, otherwise specified flash para file path: eg: chips/bl808/efuse_bootheader/flash_para.bin
flash_para = chips/bl808/efuse_bootheader/flash_para.bin
decompress_write = true
file = chips/bl808/img_create_mcu/bootinfo_group0.bin chips/bl808/img_create_mcu/img_group0.bin
address = 0 2000
```

most importantly: I dumped the flash on a factory Ox64, which was writing:
```
 |  _ \             / _|/ _|    | |     | |     | |    
 | |_) | ___  _   _| |_| |_ __ _| | ___ | | __ _| |__  
 |  _ < / _ \| | | |  _|  _/ _` | |/ _ \| |/ _` | '_ \ 
 | |_) | (_) | |_| | | | || (_| | | (_) | | (_| | |_) |
 |____/ \___/ \__,_|_| |_| \__,_|_|\___/|_|\__,_|_.__/ 

Build:19:50:39,Nov 20 2022
Copyright (c) 2022 Bouffalolab team
dynamic memory init success,heap size = 93 Kbyte 
sig1:ffff32ff
```

then wrote it back with `bflb-mcu-tool --chipname=bl808 --port=/dev/ttyACM0 --baudrate=2000000 --firmware flash-good.bin` (both with the changes listed here and without)

afterwards, there's no more messages to UART. At least this rules out bad compiler setup. Logs from flashing attached

[log.txt](https://github.com/bouffalolab/bl_mcu_sdk/files/10829856/log.txt)

gamelaster, 2023-02-25:
@DavidVentura sadly, I figured out that it is some issue with XIP or cache. Firmware runs, but crashes very early, so older firmwares worked OK (most of times).

https://github.com/bouffalolab/bl_mcu_sdk/issues/92

DavidVentura, 2023-02-25:
That's what I don't get -- I've dumped the old, good firmware that came with the board and reflashed it (both to the same and to different boards) and now it doesn't work anymore, so why would re-applying it make it break?

gamelaster, 2023-02-25:
@DavidVentura how you dumped it? It is possible that you dumped the firmware also with boot header, but bflb-mcu-tool (at least, the older one), don't check for boot header, so it will have two boot headers.

NilsBe, 2023-03-01:
> I've also applied the recommendations from @gamelaster on the `bflb_mcu_tool`, replaced the CSV with the linked one and changed the values in FLASH_CFG so that they look like
>
> ```
> [FLASH_CFG] [...]

That's exactly what I did. And in my case it worked. Of course I started the dump from 0x2000. Did you do that as well?

DavidVentura, 2023-03-01:
I'd originally dumped it starting at 0x0, but have just now dumped it starting at 0x2000 and see no difference after flashing with `bflb-mcu-tool`

All steps that I took:

- With BLDevCube 1.8.3, dump flash from a chip running `ping pong` firmware, both starting at 0x0000 and 0x2000
- Edited `bflb_mcu_tool/chips/bl808/eflash_loader/eflash_loader_cfg.ini` with `flash_id` and `flash_clock_cfg` values from this thread
- Ran `bflb-mcu-tool --chipname=bl808 --port=/dev/ttyACM0 --baudrate=2000000 --firmware X`
    - both with flash that started at 0x0000 and 0x2000

I've uploaded [the content of both flash dumps](https://github.com/bouffalolab/bl_mcu_sdk/files/10860919/both-flash-contents.zip) from a good chip (start at 0x0 and 0x2000)

Can you try to flash them and see if you get the ping-pong firmware back? If not, can you provide me with a flash & command to verify it on my chips?

DavidVentura, 2023-04-23:
@NilsBe did you have a chance to try this out?