bouffalolab/bouffalo_sdk #37: Build error for BL702

kaofishy, 2022-11-29:
When building any of the example programs with `make CHIP=bl702 BOARD=bl702dk`, the following error occurs:

```
/home/tony/bl_mcu_sdk/drivers/lhal/config/bl702/device_table.c:205:19: error: 'BFLB_DEVICE_TYPE_KYS' undeclared here (not in a function); did you mean 'BFLB_DEVICE_TYPE_AES'?
  205 |       .dev_type = BFLB_DEVICE_TYPE_KYS,
      |                   ^~~~~~~~~~~~~~~~~~~~
      |                   BFLB_DEVICE_TYPE_AES
make[3]: *** [build_out/drivers/lhal/CMakeFiles/lhal.dir/build.make:368: build_out/drivers/lhal/CMakeFiles/lhal.dir/config/bl702/device_table.c.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:351: build_out/drivers/lhal/CMakeFiles/lhal.dir/all] Error 2
```

This error does not occur for any other target (BL602/BL616/BL808).

gamelaster, 2022-11-29:
That's strange, I'm using mcu sdk with BL706 just fine. I will take a look.

kaofishy, 2022-11-29:
@gamelaster the last commit that builds is 3c19b600a0cb8f618b448e2514515dbed5da8a8e , anything later gives me that error.

gamelaster, 2022-11-29:
@kaofishy ah then makes sense, I'm on older commit. Well, just rollback to the working commit, everything should work fine.

sakumisu, 2022-11-29:
Has fixed.

closed 2022-11-30 by kaofishy