bouffalolab/bouffalo_sdk #39: Datasheets not found

a-v-s, 2022-12-03:
The readme says : " Chip Reference Manual and Chip Data Manual are listed on [document](https://dev.bouffalolab.com/document) " 

When I go to the page, and then click a datasheet download link, it says "Sorry, the page is not found"

sakumisu, 2022-12-04:
Only BL602 and BL702 are available. You can find BL808 from [bl_docs](https://github.com/bouffalolab/bl_docs).

a-v-s, 2022-12-04:
Thanks for the link. It has BL602 and BL808 but no BL702. 

Either way, the link mentioned in the documentation ain't working. So should be readme be updated ?

The BL602 and BL702 datasheets and reference manuals don't download 

2 | BL702/704/706 Unclassified Datasheet |   | Datasheet | V2.6 | 2022-08-11
-- | -- | -- | -- | -- | --

Has download link https://dev.bouffalolab.com/media/upload/doc/BL702_704_706_DS_en_2.6open.pdf?token=null

4 | BL602/604 Unclassified Datasheet |   | Datasheet | V2.8 | 2022-08-03 |   |  
-- | -- | -- | -- | -- | -- | -- | --

Has download link https://dev.bouffalolab.com/media/upload/doc/BL602_BL604_DS_EN_Combo_2.8open.pdf?token=null

7 | BL702/704/706 Unclassified Reference Manual |   | Reference Manual | V1.2 | 2022-02-17
-- | -- | -- | -- | -- | --

Has download link https://dev.bouffalolab.com/media/upload/doc/BL702_BL704_706_RM_en_1.2-open.pdf?token=null

14 | BL602/604 Unclassified Reference Manual |   | Reference Manual | V1.2 | 2020-12-16 |   |  
-- | -- | -- | -- | -- | -- | -- | --

Has download link https://dev.bouffalolab.com/media/upload/doc/BL602_BL604_RM_1.2_en.pdf?token=null

sakumisu, 2022-12-05:
You should register and then you can be available.

closed 2023-01-11 by sakumisue