bouffalolab/bouffalo_sdk #40: SPI BL702 - no "Byte-inverse signal for FIFO" option - **SPI_BYTE_MSB**

pvvx, 2022-12-04:
SPI BL702 - no "Byte-inverse signal for FIFO" option?  **SPI_BYTE_MSB** 
https://github.com/bouffalolab/bl_mcu_sdk/blob/master/drivers/lhal/include/bflb_spi.h#L66

BL702/704/706 Reference Manual: 
```
spi_config: 
Address：0x4000a200
bit7 - Byte-inverse signal for each FIFO entry data
0: Byte[0] is sent out first
1: Byte[3] is sent out first
```

With SPI_BYTE_MSB is set, SPI works correctly on the test board with BL702C10.

sakumisu, 2022-12-05:
Only support 32bit, so disable it.

pvvx, 2022-12-05:
```
unsigned int SPI_ReadInvBMode(unsigned int wrdata, unsigned int size) {
	bflb_spi_feature_control(uspi0, SPI_CMD_SET_DATA_WIDTH, size);
	unsigned int regval = bflb_spi_poll_send(uspi0, wrdata);
	unsigned int i = 4 - size;
	while (i--)
		regval >>= 8;
	return regval;
}
```

closed 2023-03-06 by sakumisue