bouffalolab/bouffalo_sdk #41: Documentation about network subsystem

mindstorm38, 2022-12-12:
Hi, I'm wondering where should I find a documentation about all the network subsystem in details, I cannot find any information about Wifi/Zigbee/BT either in the reference manual or the datasheet, they are only mentioned but register/IRQ description is available.

It seems intentional, so am I missing something in another doc?

gamelaster, 2022-12-13:
Hi, those technologies are mostly under NDA, thus only way of getting more info is by reverse engineering.

mindstorm38, 2022-12-13:
Wow ok this sounds a bit difficult xD Thank you anyway 👍

gamelaster, 2022-12-13:
@mindstorm38 oh sorry! I was in assumption this is directly BL808 question, but it might be not.
About what information are you interested? If about registers and other stuff, then it is situation described in my previous message. If it is about how to get Wifi/Zigbee/BT working at all, BL60X and BL70X are supported in BL_IOT_SDK and BL_MCU_SDK, and support for BL808 is not officially released yet.

mindstorm38, 2022-12-13:
> If about registers and other stuff, then it is situation described in my previous message. If it is about how to get Wifi/Zigbee/BT working at all, BL60X and BL70X are supported in BL_IOT_SDK and BL_MCU_SDK, and support for BL808 is not officially released yet.

Ok, I can start by looking at BL60X and BL70X to understand how they work, but without registers I will not be able to do anything I guess.

sakumisu, 2022-12-13:
Registers are not available, all of iot business are the same, just work with applications, no think about hardware.

mindstorm38, 2022-12-13:
I would have liked to build a custom application on that, no problem tho.

closed 2022-12-13 by mindstorm38