bouffalolab/bouffalo_sdk #42: (BL808) Incoherency between datasheet memory mapping and linker script

mindstorm38, 2022-12-13:
https://github.com/bouffalolab/bl_mcu_sdk/blob/16d3a819c86dab25e3c20155a3563c2dc9bc7220/bsp/board/bl808dk/bl808_flash_d0.ld#L25-L30

https://github.com/bouffalolab/bl_mcu_sdk/blob/16d3a819c86dab25e3c20155a3563c2dc9bc7220/bsp/board/bl808dk/bl808_flash_m0.ld#L25-L30

These lines of the BL808 linker script doesn't correspond at all to the datasheet (page 12-13, at https://raw.githubusercontent.com/bouffalolab/bl_docs/main/BL808_DS/en/BL808_DS_1.2_en.pdf):

[image]

[image]

Could you explain why some of these values are correct (XIP, XRAM), the most annoying values are for D0 RAM, where it is set to `0x3EFF8000` in linker script, but `0x3EF80000` (notice the shift) in datasheet, values for M0 are correct. I'm also wondering where TCM is specified in the memory mapping tables?

*Note: I'm posting this issue here, but this is related to https://github.com/bouffalolab/bl_docs, but there is not much activity there.*

sakumisu, 2022-12-15:
448K of DRAM is used for media peripherals default, so only 64K is used for users, including vram, the result is 28+4+32+32.

mindstorm38, 2022-12-15:
This make sense, thank you! It would be nice to have the TCM documented, I can't find any information about it in DS/RM.

sakumisu, 2022-12-15:
No tcm, just make part of ocram to be tcm for code and data.

mindstorm38, 2022-12-15:
Thank you, much clearer now 👍

closed 2022-12-15 by mindstorm38