bouffalolab/bouffalo_sdk #44: JTAG support on the Pine64 0x64

Fishwaldo, 2022-12-19:
Hi,
I"m trying to get a JLINK programmer running with the ox64. 

First issue I see is - the JTAG pins on the ox64 are also the same pins used for UART0 and UART1. 

Is there a way to remap the JTAG pins to another GPIO (maybe via efuse?)

2nd - even with a empty flash - I can't get JTAG working. (tried both in Bootloader mode, and normal mode with empty flash - which wouldn't make sense as those pins are used for flashing)

(I see this in the board BSP - https://github.com/bouffalolab/bl_mcu_sdk/blob/16d3a819c86dab25e3c20155a3563c2dc9bc7220/bsp/board/bl808dk/board.h#L6 but there is no actual function in the SDK)

sakumisu, 2022-12-19:
Please use the default jtag pin to have a try. See this [default jtag pin](https://github.com/bouffalolab/bl_mcu_sdk/tree/master/tools/cklink_firmware)

Fishwaldo, 2022-12-22:
I've made more progress - I can connect now when the board is in boot mode, but when running my app, JTAG wont connect. I understand I probably have to set the pins in my code? Any pointers where to do that?

This is what I've tried:
```
        struct bflb_device_s *gpio = bflb_device_get_by_name("gpio");
        bflb_gpio_init(gpio, GPIO_PIN_12, GPIO_DRV_0 | GPIO_FLOAT | GPIO_SMT_EN | GPIO_FUNC_JTAG_M0);
        bflb_gpio_init(gpio, GPIO_PIN_13, GPIO_DRV_0 | GPIO_FLOAT | GPIO_SMT_EN | GPIO_FUNC_JTAG_M0);
        bflb_gpio_init(gpio, GPIO_PIN_6, GPIO_DRV_0 | GPIO_FLOAT | GPIO_SMT_EN | GPIO_FUNC_JTAG_M0);
        bflb_gpio_init(gpio, GPIO_PIN_7, GPIO_DRV_0 | GPIO_FLOAT | GPIO_SMT_EN | GPIO_FUNC_JTAG_M0);
```
I also tried (Found in sipeed repo):
```
        GLB_GPIO_Cfg_Type gpio_cfg;

        gpio_cfg.drive = 0;
        gpio_cfg.smtCtrl = 1;
        gpio_cfg.pullType = GPIO_PULL_NONE;

        gpio_cfg.gpioMode = GPIO_MODE_AF;
        gpio_cfg.gpioFun = GPIO_FUN_JTAG_M0;
        gpio_cfg.gpioPin = GLB_GPIO_PIN_12;
        GLB_GPIO_Init(&gpio_cfg);
        gpio_cfg.gpioPin = GLB_GPIO_PIN_13;
        GLB_GPIO_Init(&gpio_cfg);
        gpio_cfg.gpioPin = GLB_GPIO_PIN_6;
        GLB_GPIO_Init(&gpio_cfg);
        gpio_cfg.gpioPin = GLB_GPIO_PIN_7;
        GLB_GPIO_Init(&gpio_cfg);
```

DebugServer Output: (when running normal firmware, not in BootLoader Mode)
```
CONNECT: Start to connect target (Enter target_open).
CONNECT: Detect JTAG port for RISC-V cores.
CONNECT: 	Configure cJTAG with 2-wires
CONNECT: 	Configure CDI type to 2-wires.
CONNECT: 	Read IDCODE Gets 0x0, Invalid IDCODE.
CONNECT: 	Configure CDI type to 5-wires.
CONNECT: 	Read IDCODE Gets 0xffffffff, Manufid is 0x7ff.
CONNECT: Can't detect JTAG Port for RISC-V DM,keep going
CONNECT: Check the DEBUG ARCH automatically.
CONNECT: 	Read IDCODE Gets 0xffffffff, Manufid is 0x7ff.
CONNECT: Try to connect target and get arch_ops.
CONNECT: +--Sync hacr width between HAD and ICE(hacr_8 default).--+
CONNECT: 	Set HACR width to 8 according to program arguments.
CONNECT: +--Configure the channel between HAD and ICE.--+
CONNECT: Configure CDI type.
CONNECT: 	Configure link with CDI 2.
CONNECT: 	Read HAD reg 0x1f get 0x0.
CONNECT: 	0x0 is not an effective HID, target is identified as CDI 5.
CONNECT: Configure HACR width.
CONNECT: 	Configure link with hacr_width_16.
CONNECT: 	Read HID get 0xffffffff(not an effective HID).
CONNECT: 	Target is identified as HACR width 8.
T-HEAD: CKLink_Lite_V2, App_ver unknown, Bit_ver null, Clock 2526.316KHz,
       5-wire, With DDC, Cache Flush On, SN CKLink_Lite_Vendor-rog 595D5C.
CONNECT: +--Attempt to obtain multi-cores and multi-processors data.--+
CONNECT: Check multi processors information.
CONNECT: 	Try to check whether the target is a MCPU with HACR_16.
CONNECT: 	Set HACR width to 16 according to program arguments.
CONNECT: 	Set HACR width to 8 according to program arguments.
CONNECT: 	Target is not multi-processors as it has no HAD_TOP.
CONNECT: Check multi-cores (CK860/C860) information.
CONNECT: 	Target is not multi-cores as hacr_width is 8.
CONNECT: Check multi-cores (CK910/C910) information.
CONNECT: 	Target is not multi-cores as hacr_width is 8.
CONNECT: +--Try to access CPU.--+
CONNECT: Init hacr width for CPU 0.
CONNECT: 	Set cpus[i].hacr_width with had_info->hacr_width 0CONNECT: Check HAD connection for CPU 0.
CONNECT: 	Read HID get 0xffffffff, HAD version is HAD_V5.
CONNECT: 	Write DDCADDR with 0x00000000000abcd0.
CONNECT: 	Read  DDCADDR get  0xffffffffffffffff.
CONNECT: 	Can not access to HAD module as DDCADDR can not be written.
Extra Info: hacr width is 8.
ERROR: Fail to check had module of CPU 0, please check:
	1. the physical connection;
	2. the power of the target;
	3. the link clock (link clock <= (CPU clock/2));
	4. the reference voltage of link, default 3.3v;
For more information, please run Debugserver with connect log:
	 Console: run Debugserver with arguments "--debug connect".
	 UI     : select "Connect" from menu "Setting->Verbose Setting".
```

sakumisu, 2022-12-22:
If you use `bflb_gpio_init`, you need add `GPIO_MODE_ALTERNATE`. Did you use mcu sdk demo or dual core demo in sipeed sdk?

sakumisu, 2022-12-22:
Which demo do you use?

Fishwaldo, 2022-12-22:
I solved the issue. Its clashing with the bl808dk setup. (specifically the board_adc_gpio_init). 

Would you accept a PR to add the Pine64 ox64 as a board/bsp?

sakumisu, 2022-12-23:
Sorry, our board is just reference, so if you need to modify, recommend you to use your own board config excluding sdk.

sakumisu, 2022-12-23:
You can build with `BOARD_DIR=xxxx` to use your board.

closed 2022-12-28 by Fishwaldo