bouffalolab/bouffalo_sdk #46: BLE and WiFi support in MCU SDK 2.0?

kaofishy, 2022-12-24:
Hi there,

Are there any plans to support BLE and WiFi like version 1.4.5 and earlier did? I'm specifically interested in both BL702 and BL808. I see that BL808 is only supported in 2.0, so is there no official way to develop BLE/WiFi firmware for BL808?

Thanks,
Tony

gamelaster, 2022-12-24:
AFAIK MCU SDK supported only BLE.

kaofishy, 2022-12-24:
> AFAIK MCU SDK supported only BLE. 

You're right, it was BLE only. The issue is that BLE is missing from the "new" MCU SDK.

gamelaster, 2022-12-24:
@kaofishy correct. MCU SDK seems to be only for MCU usage, so even I was surprised to see BLE support. I think RF protocols will be supported only in IOT sdk

kaofishy, 2022-12-24:
If that is truly the case then it just seems like unnecessary segmentation, it's not like the IoT SDK is under an NDA or anything, just adding confusion.

sakumisu, 2022-12-24:
Please wait! All the wireless protocols will support. It is in refactoring.

sakumisu, 2022-12-24:
You can get some secrets in [CMakeLists](https://github.com/bouffalolab/bl_mcu_sdk/blob/master/CMakeLists.txt)

kaofishy, 2022-12-24:
@sakumisu oh that's great to hear, thanks for the quick reply and keep up the good work, the new API is much nicer to work with than the old one.

kamushadenes, 2023-01-29:
Hello folks, any ETA on the WiFi support?

sakumisue, 2023-03-31:
done

andrewleech, 2023-03-31:
Does this mean wifi and/or ble support is now available on BL808?
If so could you please link to some details?

sakumisu, 2023-03-31:
> Does this mean wifi and/or ble support is now available on BL808? If so could you please link to some details?

Readme has noted, only for bl616/bl618

andrewleech, 2023-03-31:
I saw that and was hoping it was out of date... this issue was raised for bl702 and bl808, if their support is still pending the issue should be left open!

Fishwaldo, 2023-05-18:
Hey @sakumisu is there still plans to release wifi for the BL808?

sfranzyshen, 2023-11-19:
@sakumisu why can't the bl808 community ever get any answers about BLE and WiFi support ON THE BL808 platform???

Will wifi/ble  "EVER" be an option for the bl808 platform ??? ... 

I am not asking "WHEN" ... I am asking "IF EVER" ???

can you at least answer that??

gamelaster, 2023-11-20:
@sfranzyshen there is wifi support on the BL808 platform. Bouffalo provided blobs both for Linux (those are fresh ones), and then ones are in IoT SDK.

sfranzyshen, 2023-11-20:
> @sfranzyshen there is wifi support on the BL808 platform. Bouffalo provided blobs both for Linux (those are fresh ones), and then ones are in IoT SDK.

I am asking about HERE.

sfranzyshen, 2024-01-19:
still nothing ... what a shame