bouffalolab/bouffalo_sdk #47: Flashing issues on BL808 (on Ox64)

mindstorm38, 2022-12-24:
Hi, I'm trying to flash the BL808 used on the Ox64 board (20221018 version) and I have many questions about it:

1) Do you provide any support for flashing it using JFlash (JLink)? I've been able to read the memory using JMem, working well, and I can even read registers (the cpu id one for example). But when trying to flash it doesn't work, I guess because it's QSPI and read-only when using memory-mapped access... So I'll be forced to use an Open FlashLoader binary if I want it to work, and I was wondering if you provide such things or workarounds...

2) When using BLDevCube with JLink mode, I can see/select the JLink port that is connected to the BL808 and click "Create & Download", and it doesn't go that far as you can see in the logs:
```
...
[17:18:26.264] - Write flash img
[17:18:26.266] - Image hash is b'e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855'
[17:18:26.267] - Header crc: b'77f215f5'
[17:18:26.268] - Write flash img
[17:18:26.270] - Encrypt efuse data
[17:18:26.272] - ========= eflash loader config =========
[17:18:26.287] - Version: eflash_loader_v2.4.3
[17:18:26.288] - Program Start
[17:18:26.289] - ========= eflash loader cmd arguments =========
[17:18:26.289] - None
[17:18:26.289] - Config file: D:\Programmes\BLDevCube-1.8.1\chips\bl808\eflash_loader\eflash_loader_cfg.ini
[17:18:26.290] - serial port is 175001652
[17:18:26.290] - cpu_reset=False
[17:18:26.290] - chiptype: bl808
[17:18:26.290] - ========= Interface is JLink =========
[17:18:26.291] - Bootrom load
[17:18:26.291] - ========= get_boot_info =========
[17:18:26.291] - interface not fit
[17:18:26.291] - ErrorCode: 0050, ErrorMsg: BFLB IMG LOAD SHAKEHAND FAIL
[17:18:26.291] - Burn Retry
[17:18:26.291] - 0
[17:18:26.291] - Burn return with retry fail
```

3) When flashing using UART on BLDevCube (I push the boot button before powering on the chip), the handshake is successful, the image is built, but I get an error when it tries to enter xip mode (see logs below). It seems that flashing is successful, but when I use JMem (JLink) to read the flash at 0x58000000, it's filled with 0xFF, so I'm wondering if the flashing actually worked (maybe rollback?).
*Note that I'm playing with Rust and linker scripts (all based on your SDK), if you think that it comes from this I'll try on the sdk, but I'd be really surprised that my binary (manually verified to contain right startup instructions) would prevent flashing success.*

```
[17:08:59.756] - EFUSE_CFG
[17:08:59.757] - BOOTHEADER_GROUP0_CFG
[17:08:59.758] - BOOTHEADER_GROUP1_CFG
[17:08:59.779] - Create bootheader using D:\Programmes\BLDevCube-1.8.1\chips\bl808\img_create_mcu/efuse_bootheader_cfg.ini
[17:08:59.780] - Updating data according to <D:\Programmes\BLDevCube-1.8.1\chips\bl808\img_create_mcu/efuse_bootheader_cfg.ini[BOOTHEADER_GROUP0_CFG]>
[17:08:59.784] - Created file len:352
[17:08:59.789] - Create bootheader using D:\Programmes\BLDevCube-1.8.1\chips\bl808\img_create_mcu/efuse_bootheader_cfg.ini
[17:08:59.790] - Updating data according to <D:\Programmes\BLDevCube-1.8.1\chips\bl808\img_create_mcu/efuse_bootheader_cfg.ini[BOOTHEADER_GROUP1_CFG]>
[17:08:59.795] - Created file len:352
[17:08:59.798] - Create efuse using D:\Programmes\BLDevCube-1.8.1\chips\bl808\img_create_mcu/efuse_bootheader_cfg.ini
[17:08:59.799] - Updating data according to <D:\Programmes\BLDevCube-1.8.1\chips\bl808\img_create_mcu/efuse_bootheader_cfg.ini[EFUSE_CFG]>
[17:08:59.804] - Created file len:256
[17:08:59.811] - Image create path: D:\Programmes\BLDevCube-1.8.1\chips\bl808\img_create_mcu
[17:08:59.812] - Config file: D:\Programmes\BLDevCube-1.8.1\chips\bl808\img_create_mcu/img_create_cfg.ini
[17:08:59.814] - Image hash is b'4642d564bc85498d292f113851b9a92c9780a838f623766099bd8a30a06d6eff'
[17:08:59.830] - Header crc: b'121ba109'
[17:08:59.831] - Write flash img
[17:08:59.834] - Image hash is b'e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855'
[17:08:59.835] - Header crc: b'77f215f5'
[17:08:59.835] - Write flash img
[17:08:59.837] - Encrypt efuse data
[17:08:59.839] - ========= eflash loader config =========
[17:08:59.854] - Version: eflash_loader_v2.4.3
[17:08:59.855] - Program Start
[17:08:59.856] - ========= eflash loader cmd arguments =========
[17:08:59.856] - None
[17:08:59.856] - Config file: D:\Programmes\BLDevCube-1.8.1\chips\bl808\eflash_loader\eflash_loader_cfg.ini
[17:08:59.858] - serial port is COM4
[17:08:59.858] - cpu_reset=False
[17:08:59.858] - chiptype: bl808
[17:08:59.858] - ========= Interface is uart =========
[17:08:59.859] - com speed: 115200
[17:08:59.859] - Bootrom load
[17:08:59.859] - ========= get_boot_info =========
[17:08:59.859] - ========= image get bootinfo =========
[17:09:00.196] - tx rx and power off, press the machine!
[17:09:00.197] - cutoff time is 0.05
[17:09:00.257] - power on tx and rx
[17:09:01.262] - reset cnt: 0, reset hold: 0.05, shake hand delay: 0.1
[17:09:01.263] - clean buf
[17:09:01.266] - send sync
[17:09:01.799] - ack is b'4f4b'
[17:09:01.847] - shake hand success
[17:09:02.355] - data read is b'01000808000000000b14c102e9f12bcf0eb455002ff4fb18'
[17:09:02.356] - ========= ChipID: b40ecf2bf1e9 =========
[17:09:02.356] - Get bootinfo time cost(ms): 2497.1748046875
[17:09:02.356] - clock para file: D:\Programmes\BLDevCube-1.8.1\chips/bl808/efuse_bootheader/clock_para.bin
[17:09:02.357] - change bdrate: 115200
[17:09:02.357] - Clock PLL set
[17:09:02.357] - clock para:
[17:09:02.357] - 504346470404000003010300010200020101000101010101c0c4e229
[17:09:02.362] - Set clock time cost(ms): 5.013427734375
[17:09:02.493] - Read mac addr
[17:09:02.498] - flash set para
[17:09:02.499] - get flash pin cfg from bootinfo: 0x04
[17:09:02.499] - set flash cfg: 14104
[17:09:02.499] - Set flash config
[17:09:02.504] - Set para time cost(ms): 4.98779296875
[17:09:02.505] - ========= flash read jedec ID =========
[17:09:02.509] - Read flash jedec ID
[17:09:02.509] - readdata:
[17:09:02.510] - b'ef601800'
[17:09:02.510] - Finished
[17:09:02.514] - Program operation
[17:09:02.515] - Dealing Index 0
[17:09:02.515] - ========= programming D:\dev\projects\bflb-rust-experiment\app.bin to 0x58000000
[17:09:02.519] - flash para file: D:\Programmes\BLDevCube-1.8.1\chips/bl808/efuse_bootheader/flash_para.bin
[17:09:02.519] - Set flash config
[17:09:02.533] - Set para time cost(ms): 13.519287109375
[17:09:02.533] - ========= flash load =========
[17:09:02.534] - ========= flash erase =========
[17:09:02.534] - Erase flash from 0x58000000 to 0x58000733
[17:09:02.540] - Erase time cost(ms): 6.013427734375
[17:09:02.706] - Load 1844/1844 {"progress":100}
[17:09:02.707] - Load 1844/1844 {"progress":100}
[17:09:02.707] - Write check
[17:09:02.713] - Flash load time cost(ms): 171.0556640625
[17:09:02.714] - Finished
[17:09:02.714] - Sha caled by host: a5a16e48a85b7f152a808103c63a3eeb34bc1b6ad8bf02932225b1498bd550fe
[17:09:02.714] - xip mode Verify
[17:09:04.724] - ack is b''
[17:09:04.725] - Not ack OK
[17:09:04.725] - Retry
[17:09:06.739] - ack is b''
[17:09:06.740] - Not ack OK
[17:09:06.740] - Retry
[17:09:08.752] - ack is b''
[17:09:08.753] - Not ack OK
[17:09:08.753] - Retry
[17:09:10.756] - ack is b''
[17:09:10.757] - Not ack OK
[17:09:10.757] - Read Fail
[17:09:12.758] - ack is b''
[17:09:12.759] - ErrorCode: 0039, ErrorMsg: BFLB FLASH XIP MODE ENTER FAIL
[17:09:12.759] - a bytes-like object is required, not 'NoneType'
Traceback (most recent call last):
File "libs\bflb_eflash_loader.py", line 3420, in efuse_flash_loader_do
File "libs\bflb_eflash_loader.py", line 2045, in flash_load_specified
File "libs\bflb_eflash_loader.py", line 1948, in flash_load_opt
TypeError: a bytes-like object is required, not 'NoneType'
[17:09:12.760] - Burn Retry
[17:09:12.760] - 0
[17:09:12.760] - Burn return with retry fail
```

Thank you!

gamelaster, 2022-12-24:
What USB UART convertor do you use?

mindstorm38, 2022-12-24:
> What USB UART convertor do you use? 

CP2102

Fishwaldo, 2022-12-25:
Erase flash from 0x58000000 to 0x58000733

This is a weird address compared to what I see when normally flashing. What command line options are you using?

The 0x58xxxxxx range is the "mapped" XIP range (same space for each core) which I guess is setup in the bootrom. I believe the physical address space is around 0x2000 for the M0 core. ( and 0x52000 for D0 core from memory - Have a look around the eflash config files for other hints).

sakumisu, 2022-12-25:
Jlink mode in devcube only support 702.

sakumisu, 2022-12-25:
Flash with Jlink need Segger support. So recommend you to use uart. Also , your uart2ttl module need to be ok then you can flash success.

gamelaster, 2022-12-25:
I didn't had success with CP2102. Try slower baudrate, but if it will not work, I recommend trying another USB uart dongle

mindstorm38, 2022-12-25:
> The 0x58xxxxxx range is the "mapped" XIP range (same space for each core) which I guess is setup in the bootrom. I believe the physical address space is around 0x2000 for the M0 core. ( and 0x52000 for D0 core from memory - Have a look around the eflash config files for other hints).

This was the issue, thank you, it works now. I still have a lot of things to learn, but this already has helped me a lot.

> Flash with Jlink need Segger support. So recommend you to use uart. Also , your uart2ttl module need to be ok then you can flash success.

I think that it's possible to write a flashloader based on the SDK using flash APIs, but it's a bit too much for now.

mindstorm38, 2022-12-25:
I'm definitely missing something here, even if it works… Because my code is linked so that the first instruction of the binary file is at 0x58000000, so when I place it at 0x2000, the first instruction is located at 0x58002000, am I missing something?

sakumisu, 2022-12-25:
No,you are wrong, 0x2000 is flash physical addr not xip addr , when you flash into 0x2000, it is 0x58000000.

mindstorm38, 2022-12-25:
Ok, so each core have the address 0x58000000 mapped to the physical address we set in BLDevCube?

sakumisu, 2022-12-26:
> 

Right

mindstorm38, 2022-12-26:
[image]

Thank you! I close this issue for now as all questions were answered.

closed 2022-12-26 by mindstorm38