bouffalolab/bouffalo_sdk #48: IPC Souce Missing

Fishwaldo, 2022-12-29:
Hi,
we have the header bl808_ipc.h present, but the source for IPC is missing. I found a copy in the bl_linux repo, but wondering if it could be added to this repo?

https://github.com/bouffalolab/bl808_linux/blob/main/bl_mcu_sdk_bl808/drivers/bl808_driver/std_drv/src/bl808_ipc.c

sakumisu, 2022-12-29:
It is in refactoring not used now, so removed.

Fishwaldo, 2022-12-30:
any ETA?

sakumisu, 2022-12-30:
Do you prepare to port with rpmsg or openmap? Maybe uncertain time.

Fishwaldo, 2022-12-31:
Using rpmsg-lite.

Fishwaldo, 2023-01-11:
if you API is not completed yet, can I make a suggestion. 
In the old implementation, you divided the 32 interupt Sources equally between the 2 other CPU's (so we had 16 "sources" per CPU). 
In the new implementation, why do you reserve IPC_INT_SRC_BIT31 as a flag to indicate which CPU sent the IPC, and that gives applications 31 sources to use for signalling instead of 16.

sakumisu, 2023-01-12:
If both cpu trigger int,  one bit cannot work.