bouffalolab/bouffalo_sdk #50: Camera, MJPEG, H264 documentation

mindstorm38, 2022-12-30:
Hi! 

I feel like there is a little asymmetry between the level of documentation between Reference Manual and SDK, regarding interactions between Camera, MJPEG and H265 modules. 

For example, the module that seems responsible for accepting CSI in DVP module (using the CAM_FRONT registers) is not documented, while CAM registers are. In general these modules are well documented but lacks of information about interactions between all of them, especially CSI/CAM, CAM/MJPEG with hardware mode.

Thank you anyway, this is hard but good learning for me :)

mindstorm38, 2022-12-31:
In the root README, MJPEG is marked supported, if it's not the case it should be fixed I think, and H264 isn't in the table.

sakumisu, 2022-12-31:
Mjpeg is supported. and have case in bl702/bl616/bl808, so why it should be fixed?

sakumisu, 2022-12-31:
We will not release H264 in our sdk not, you can refer to sipeed sdk.

mindstorm38, 2022-12-31:
> Mjpeg is supported. and have case in bl702/bl616/bl808, so why it should be fixed?

This was in response to @gamelaster , I thought MJPEG wasn't supported as he said...

> We will not release H264 and CSI in our sdk now, you can refer to sipeed sdk.We only provide common peripherals.

Ok, no problem I'll look at their SDK

sakumisu, 2022-12-31:
CAM with dvp and mjpeg are supported in BL702/BL616/BL808. If you want to use h264 and csi, just copy drivers from sipeed sdk to mcu sdk, we do not use now so removed them not compile them.

sakumisu, 2022-12-31:
Its codestyle need to be refactored. So i removed them.

closed 2023-01-02 by mindstorm38