bouffalolab/bouffalo_sdk #51: UART3 HAL?

Fishwaldo, 2023-01-03:
Hi,
I see UART3 is missing from the hal driver. Any plans to update it?

Can you share the "magic" to get both TX/RX on D0 running with UART3?

(I see in the board.c only TX is setup, does this commented out line turn on RX? - https://github.com/bouffalolab/bl_mcu_sdk/blob/a574195a4b3399341794971a2ab4b8b17f4b70d7/bsp/board/bl808dk/board.c#L215)

sakumisu, 2023-01-03:
UART3 driver is the same with others, just pinmux different.

sakumisu, 2023-01-03:
UART3 function must use 21, and uart3 has irq in d0, so it is convenient to use in D0.

sakumisu, 2023-01-03:
> 

Right, just turn on it and rx will works.

Fishwaldo, 2023-01-03:
There are no GPIO_UART_FUNC_UART3_RX/GPIO_UART_FUNC_UART3_RX defines though.

Fishwaldo, 2023-01-03:
> UART3 function must use 21, and uart3 has irq in d0, so it is convenient to use in D0.

yep, I'm working on D0 right now. so 21 works for both RX/TX?

sakumisu, 2023-01-03:
> There are no GPIO_UART_FUNC_UART3_RX/GPIO_UART_FUNC_UART3_RX defines though.

Its pinmux is not the same with other uarts.

sakumisu, 2023-01-03:
uart3 pinmux is settled but uart0~uart2 can use in all pins.

Fishwaldo, 2023-01-03:
ok. Thanks

sakumisu, 2023-01-03:
Use 21 means use uart3, and pin8 must be tx, pin9 must be rx.

Fishwaldo, 2023-01-03:
> Use 21 means use uart3, and pin8 must be tx, pin9 must be rx.

we can't use pin8/9 on the ox64 as they are connected to the USB-C connector. So can I just use a different GPIO_PIN_X ?

sakumisu, 2023-01-03:
To prevent misleading initialization of other uart gpio's with use ``bflb_uart_gpio_init``, so hardcode this.

sakumisu, 2023-01-03:
pin is a multiple of 4 will be tx, and pin+1 will be rx.

sakumisu, 2023-01-03:
Such as 0-1, 4-5, 8-9...

Fishwaldo, 2023-01-03:
> To prevent misleading initialization of other uart gpio's with use `bflb_uart_gpio_init`, so hardcode this.

you mean with 21:

```
bflb_uart_gpio_init(gpio, 21, GPIO_PIN_X);
bflb_uart_gpio_init(gpio, 21, GPIO_PIN_Y);
```

sakumisu, 2023-01-03:
No no no, to prevent users to use this with uart3.

sakumisu, 2023-01-03:
Only uart0~uart2 can use this api.

Fishwaldo, 2023-01-03:
so I can just use:
```
 bflb_gpio_init(gpio, GPIO_PIN_16, 21 | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
 bflb_gpio_init(gpio, GPIO_PIN_17, 21 | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
```

sakumisu, 2023-01-03:
Right.

Fishwaldo, 2023-01-03:
got it :)

Fishwaldo, 2023-01-03:
Just for clarity - in the Datasheet, MM_UART_XX is uart3?

so technically I would have to use gpio 15/16 instead of 16/17?

sakumisu, 2023-01-03:
No 15 is not tx.

Fishwaldo, 2023-01-03:
Sorry, got confused with physical pin versus GPIO pin in the datasheet.

closed 2023-01-11 by sakumisue