bouffalolab/bouffalo_sdk #53: BL616 Zigbee?

justchen1369, 2023-01-06:
Hi, I can't seem to find any documentation about how to use Zigbee on the upcoming BL616 chip. Are there any resources available about that?

sakumisue, 2023-01-11:
In NDA can be available.

justchen1369, 2023-01-11:
Does that mean that it's potentially accessible or hobbyists, or perhaps there will be a public release at some point?

3vlig, 2023-03-14:
any progress here? or can a hobbyist/private user get this even when under NDA?

Opisek, 2024-04-02:
Any progress? The chips are explicitly advertised as Zigbee-capable, yet there seems to be no way to actually get it to work.

Virus-V, 2026-05-13:
Hi, Zigbee support for BL616 is currently available only under NDA. The public SDK does not include Zigbee resources.

If you need Zigbee functionality, please reach out to Bouffalo Lab through official channels to discuss NDA arrangements.

Closing this as it's not a public SDK issue. Thanks for your interest.