bouffalolab/bouffalo_sdk #54: SDH Interupts

Fishwaldo, 2023-01-09:
Hi,
Can you confirm that the SDH peripherial is also only hooked up to the M0/LP core and not D0 core?

also, can you tell us what these interupts are?
```
SDH_MMC1_IRQn = IRQ_NUM_BASE + 32,        /*!< SDH MMC1 Interrupt                                                /
    SDH_MMC3_IRQn = IRQ_NUM_BASE + 33,        /!< SDH MMC3 Interrupt                                                /
    SDH2PMU_WAKEUP1_IRQn = IRQ_NUM_BASE + 34, /!< SDH2PMU Wakeup1 Interrupt                                         /
    SDH2PMU_WAKEUP3_IRQn = IRQ_NUM_BASE + 35, /!< SDH2PMU Wakeup3 Interrupt
```

does this mean that MMC2 IRQ is the M0 IRQ and the others are routed to the D0 core?

sakumisu, 2023-01-11:
These are typos, not removed.

sakumisu, 2023-01-11:
Just ignore.

closed 2023-01-11 by Fishwaldo