bouffalolab/bouffalo_sdk #55: bflb_mtimer_delay_us and bflb_mtimer_delay_ms broken

alexhorner, 2023-01-09:
https://github.com/bouffalolab/bl_mcu_sdk/blob/769349f292851c458c2702e62e1fb777d2f507ce/drivers/lhal/src/bflb_mtimer.c#L72
https://github.com/bouffalolab/bl_mcu_sdk/blob/769349f292851c458c2702e62e1fb777d2f507ce/drivers/lhal/src/bflb_mtimer.c#L80

Both `bflb_mtimer_delay_us` and `bflb_mtimer_delay_ms` do not work. Placing something in the while loop, such as `printf("")`, fixes them. It appears they may be being optimised away by the compiler?

Right now, the SDK produces no output on the UART when using them without modification

sakumisu, 2023-01-10:
No, it works in ours.

alexhorner, 2023-01-10:
> No, it works in ours.

When recloning the SDK repo, for me it still does not work. Turning optimisations to `-O0` fixes

Fishwaldo, 2023-01-11:
hi,
I have occasionally hit the same problem. Today I got a backtrace:
```
(gdb) bt full
#0  bflb_mtimer_get_time_us () at /home/fish/bl808/bl_mcu_sdk/drivers/lhal/include/arch/risc-v/t-head/Core/Include/core_rv64.h:775
        tmp_low = 0
        tmp_high = 0
        tmp_low1 = 0
        tmp_high1 = 0
#1  bflb_mtimer_get_time_ms () at /home/fish/bl808/bl_mcu_sdk/drivers/lhal/src/bflb_mtimer.c:69
No locals.
#2  bflb_mtimer_delay_ms (time=time@entry=100) at /home/fish/bl808/bl_mcu_sdk/drivers/lhal/src/bflb_mtimer.c:84
        start_time = 0
```

It appears the mtimer occasionally wont get started. bflb_mtimer_get_time_us() is always returning 0?

Fishwaldo, 2023-01-11:
by the way, this is on D0 core (at least in my back trace)

Fishwaldo, 2023-01-11:
as a side note - ```CPU_MTimer_Delay_MS(100)``` seems to work flawlessly. while we get random failures with the bflb_mtimer* functions

sakumisu, 2023-01-14:
I have tested with sdk, it still works ok, so which board and which demo do you use?

alexhorner, 2023-01-14:
> I have tested with sdk, it still works ok, so which board and which demo do you use?

We are developing in the Ox64. When experiencing this issue, I freshly cloned the repo and compiled peripherals/gpio/gpio_input_output for M0

sakumisu, 2023-01-14:
Yes, ox64 also , with helloworld in M0, works ok. So this issue need be solved by yourself.

alexhorner, 2023-01-14:
> Yes, ox64 also , with helloworld in M0, works ok. So this issue need be solved by yourself.

Please confirm that you cloned a new repo? I will reclone again and come back with as much information as I can if it still does not work

alexhorner, 2023-01-16:
> Yes, ox64 also , with helloworld in M0, works ok. So this issue need be solved by yourself.

Hello. Yes, I have tested again, helloworld is working. I have another example instead.

`examples/peripherals/usbhost` is definitely affected by the problem. I cloned the SDK fresh and compiled it for M0. It does not work. I have attached the *broken* binary files - [usbhost_bl808_m0 Broken.zip](https://github.com/bouffalolab/bl_mcu_sdk/files/10423989/usbhost_bl808_m0.Broken.zip). I did not change any files in the SDK when making this.

I fixed it by adding `printf("");` to `bflb_mtimer_delay_us` and `bflb_mtimer_delay_ms` as described in the original issue.

I also fixed it by not putting `printf("");` but instead turning off optimisations.

Hopefully this lets you reproduce the issue, thank you

sakumisu, 2023-01-16:
```
Build:16:53:07,Jan 16 2023
Copyright (c) 2022 Bouffalolab team
=========== flash cfg ==============
jedec id   0xEF4018
mid            0xEF
iomode         0x04
clk delay      0x01
clk invert     0x01
read reg cmd0  0x05
read reg cmd1  0x35
write reg cmd0 0x01
write reg cmd1 0x31
qe write len   0x01
cread support  0x00
cread code     0xFF
burst wrap cmd 0x77
=====================================
dynamic memory init success,heap size = 155 Kbyte 
sig1:ffff32ff
sig2:0000ffff
Starting usb host task...
[I/USB] EHCI HCIVERSION:0100
[I/USB] EHCI HCSPARAMS:000001
[I/USB] EHCI HCCPARAMS:0006
```

sakumisu, 2023-01-16:
Works ok.

alexhorner, 2023-01-16:
> Works ok.

Please could you send your binaries for comparison?

sakumisu, 2023-01-16:
If you have sipeed M1S dock, i can send you.

sakumisu, 2023-01-16:
You can tell me which console pins do you use? I will build bin for you.

Zagitta, 2023-01-16:
This sounds like a classical miss compilation, what's the exact toolchain version everyone's using?

sakumisu, 2023-01-16:
> This sounds like a classical miss compilation, what's the exact toolchain version everyone's using?

Use t-head v2.6.1

sakumisu, 2023-01-16:
gcc version 10.2.0 (Xuantie-900 elf newlib gcc Toolchain V2.6.1 B-20220906)

arm000, 2023-01-17:
I've seen this instability similar to what @alexhorner describes, where SDK examples don't work, but adding an extra printf() at the start of main() causes them to start working again.  From what I can tell when running sample applications on M0, if flash group1 is not programmed, the D0 processor starts executing whatever code is in flash and sometimes hangs the system.  If I disable the calls to  GLB_Release_CPU(GLB_CORE_ID_D0) and GLB_Release_CPU(GLB_CORE_ID_LP) in board.c the system appears stable again.

closed 2023-03-06 by sakumisue