bouffalolab/bouffalo_sdk #63: BL808 running problem

elektro-NIK, 2023-01-16:
Hello! 
I have a Pine64's Ox64 board (with 2MB flash) and I don't run helloworld app.

I have:
- bl_mcu_sdk master branch
- bflb-mcu-tool 1.8.1 

What I do:
- clone bl_mcu_sdk
- copy bsp/board/bl808bk to bsp/board/Ox64
- change xip_memory LENGTH from 32M to 2M in both linker scripts
- enter into examples/helloworld
- make CHIP=bl808 CPU_ID=m0 BOARD=Ox64
- bflb-mcu-tool --chipname=bl808 --port=/dev/ttyUSB0 --baudrate=2000000 --firmware build/build_out/helloworld_bl808_m0.bin
- flash writing and verifying success
- after reboot board I have not any messages in serial monitor (GPIO14/15 2000000 baud)

I don't have JTAG adapter for debug MCU startup yet. Maybe I make some stupid mistake and someone shows me this?

gamelaster, 2023-01-16:
Hi @elektro-NIK , this is due Flash memory misconfiguration. 

Try this: https://github.com/bouffalolab/bl_mcu_sdk/issues/32#issuecomment-1345646038

elektro-NIK, 2023-01-17:
@gamelaster I try your instruction, but my board still not work.

`cat $HOME/.local/lib/python3.10/site-packages/bflb_mcu_tool/utils/flash/bl808/flashcfg_list.csv | grep ef6015`
> ef6015,W25Q64JW_ef6017.conf

`cat $HOME/.local/lib/python3.10/site-packages/bflb_mcu_tool/chips/bl808/eflash_loader/eflash_loader_cfg.ini | grep -A 4 FLASH_CFG`
> [FLASH_CFG]
> flash_id = ef6015
> #bit 7-4 flash_clock_type: 0:120M wifipll, 1:xtal, 2:128M cpupll, 3:80M wifipll, 4:bclk, 5:96M wifipll
> #bit 3-0 flash_clock_div
> flash_clock_cfg = 0x43

I see on Pine64 wiki page are wrong datasheets for flash chips: GigaDevice instead of Winbond on the board. I will try to edit this page.

gamelaster, 2023-01-17:
Hmm, that's weird. Can you please take a picture of SPI Flash chip? Thanks

elektro-NIK, 2023-01-17:
[image]

NilsBe, 2023-02-19:
I have managed to compile and flash the Hello World example on the 16MB version but it didn't work on the 2MB version. 
Do you have the 16MB version available to test that one?

elektro-NIK, 2023-02-20:
Yep, I have the same behavior. All works on 16MB and not works on 2MB board.

gamelaster, 2023-02-21:
I'm working on it. It is 100% issue with flash configuration, because BootROM can both read and write to SPI Flash.
Only when the firmware is executed, the special configuration is used to use higher speed clocks and flash related modes to have higher transfer speeds, and this is not properly configured, so flash don't work at all.