bouffalolab/bouffalo_sdk #64: BL808 D0 does not work with IPC_SYNC

kaofishy, 2023-01-16:
On Sipeed M1S Dock, when compiling for the BL808 D0 CPU none of the examples worked, JTAG revealed that the core was stuck in this IPC_SYNC loop: https://github.com/bouffalolab/bl_mcu_sdk/blob/9e189b69cbc0a75ffa170f600a28820848d56432/drivers/soc/bl808/std/startup/d0/system_bl808.c#L11-L21

Commenting out the loop resolves the issue.

kaofishy, 2023-01-16:
BTW what is special about where IPC_SYNC_ADDR1 and IPC_SYNC_ADDR2 are located (0x40000000)? The linker script labels it `xram` so is it just regular SRAM?

sakumisu, 2023-01-17:
Update code.

kaofishy, 2023-01-18:
@sakumisu updated to the latest git and still doesn't work without commenting out the IPC_SYNC loop. 

This is my DevCube settings btw if that matters:
[image]

sakumisu, 2023-01-18:
please chip erase

sakumisu, 2023-01-18:
I test it is ok in M1S

kaofishy, 2023-01-18:
Tried it on two different M1s, clean repo, with chip erase enabled in DevCube, still does not boot if I don't comment out the loop. Maybe there is a timing issue that doesn't show up on all hardware?

sakumisu, 2023-01-18:
Sorry, i dont know. Ours here are all ok.

sakumisu, 2023-01-18:
Just now.

```
Build:11:27:17,Jan 18 2023
Copyright (c) 2022 Bouffalolab team
dynamic memory init success,heap size = 59 Kbyte 
sig1:ffff32ff
sig2:0000ffff
cgen1:9f7ffffd
[F][MAIN] hello world d0
[E][MAIN] hello world d0
[W][MAIN] hello world d0
[I][MAIN] hello world d0
[F][MAIN] hello world d0
[E][MAIN] hello world d0
[W][MAIN] hello world d0
[I][MAIN] hello world d0
[F][MAIN] hello world d0
[E][MAIN] hello world d0
[W][MAIN] hello world d0
[I][MAIN] hello world d0

```

sakumisu, 2023-01-18:
Update code and have a try.

kaofishy, 2023-01-18:
@sakumisu working perfectly now

closed 2023-01-18 by kaofishy

CrazyShout, 2023-02-04:
@kaofishy Hello, I have also encountered such a problem now. Did you solve the problem just by updating the code? What are your specific steps.Thank you!