bouffalolab/bouffalo_sdk #65: usbhost sample program not working

arm000, 2023-01-17:
I'm running the usbhost sample program from the SDK on M1s dock, and I'm unable to enumerate any USB Mass Storage devices.
I've compiled the sample with  "make CHIP=bl808 BOARD=bl808dk CPU_ID=m0" and used Dev Cube to flash it to group0 slot.  The sample runs and prints this in a loop:

[00:18:46.261] - do not find /dev/sda
[00:18:46.262] - do not find /dev/input0
[00:18:47.258] - do not find /dev/ttyACM0

When I plug in a USB Mass Storage device (USB flash stick) I get the following error:
[00:18:50.513] - [31m[E/USB] Failed to enable port 1

I've tried many different USB flash drives, and they all fail the same way.  Plugging in a USB mouse does work, and I get the following messages:

[00:20:02.686] - [32m[I/USB] New full-speed device on Hub 1, Port 1 connected
[00:20:02.692] - [0m[32m[I/USB] New device found,idVendor:045e,idProduct:0745,bcdDevice:0634
[00:20:02.696] - [0m[32m[I/USB] The device has 3 interfaces
[00:20:02.697] - [0m[32m[I/USB] Enumeration success, start loading class driver
[00:20:02.698] - [0m[32m[I/USB] Loading hid class driver
[00:20:02.699] - [0m[32m[I/USB] Ep=81 Attr=03 Mps=8 Interval=04 Mult=00
[00:20:02.700] - [0m[32m[I/USB] Register HID Class:/dev/input0
[00:20:02.700] - [0m[32m[I/USB] Loading hid class driver
[00:20:02.702] - [0m[32m[I/USB] Ep=82 Attr=03 Mps=10 Interval=01 Mult=00
[00:20:02.703] - [0m[32m[I/USB] Register HID Class:/dev/input1
[00:20:02.704] - [0m[32m[I/USB] Loading hid class driver
[00:20:02.705] - [0m[32m[I/USB] Ep=83 Attr=03 Mps=32 Interval=01 Mult=00
[00:20:02.706] - [0m[32m[I/USB] Register HID Class:/dev/input2

sakumisu, 2023-01-17:
Works ok, please make sure your power is good.

```
[32m[I/USB] New full-speed device on Hub 1, Port 1 connected
[0m[32m[I/USB] New device found,idVendor:31b6,idProduct:0118,bcdDevice:0200
[0m[32m[I/USB] The device has 2 interfaces
[0m[32m[I/USB] Enumeration success, start loading class driver
[0m[32m[I/USB] Loading cdc_acm class driver
[0m[32m[I/USB] Ep=01 Attr=02 Mps=64 Interval=00 Mult=00
[0m[32m[I/USB] Ep=81 Attr=02 Mps=64 Interval=00 Mult=00
[0m[32m[I/USB] Register CDC ACM Class:/dev/ttyACM0
[0m[32m[I/USB] Loading cdc_data class driver
[0mrecv over:20
0xfe 0x03 0x03 0x00 0x00 0x40 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
send over:8
0x00 0x00 0x00 0x00 0x36 0x36 0x08 0x14 0x4a 0x32 0x1a 0x30 0x0a 0x12 0x31 0x31 0x39 0x39 0x31 0x35 0x30 0x30 0x30 0x37 0x31 0x31 0x32 0x44 0x35 0x43 0x30 0x42 0x12 0x0a 0x33 0x52 0x4e 0x4c 0x30 0x34 0x30 0x55 0x5f 0x31 0x22 0x0e 0x41 0x32 0x42 0x5a 0x42 0x51 0x56 0x36 0x34 0x35 0x58 0x47 0x35 0x44 nbytes:60
```

sakumisu, 2023-01-17:
Just check your power, thanks.

arm000, 2023-01-17:
I tried as you said and and connected to my DC bench power supply, it still fails in the same way.  All USB Mass Storage devices fail to be detected with "Failed to enable port 1" error

https://drive.google.com/file/d/1Mw2h0y8q-A81zrDW_sotAS62tHtPhiqh/view?usp=sharing

sakumisu, 2023-01-18:
Maybe m1s dock usb circuit has some issue. Use our own board is ok.

```
[I/USB] New high-speed device on Hub 1, Port 1 connected
[I/USB] New device found,idVendor:0781,idProduct:5591,bcdDevice:0100
[I/USB] The device has 1 interfaces
[I/USB] Enumeration success, start loading class driver
[I/USB] Loading msc class driver
[I/USB] Get max LUN:1
[I/USB] Ep=81 Attr=02 Mps=512 Interval=00 Mult=00
[I/USB] Ep=02 Attr=02 Mps=512 Interval=00 Mult=00
[I/USB] Capacity info:
[I/USB] Block num:60062500,block size:512
[I/USB] Register MSC Class:/dev/sda
do not find /dev/ttyACM0

fa 31 c0 8e d8 8e c0 8e d0 bc 00 7c fb fc 89 e6 
bf 00 06 b9 00 01 f3 a5 ea dc 06 00 00 10 00 01 
00 00 7c 00 00 00 00 00 00 00 00 00 00 80 3f 00 
ff 00 ff 03 1e 0e 1f 3a 16 10 00 74 06 1f ea 36 
e7 00 f0 3d fb 54 75 05 8c d8 fb eb 1d 80 fc 08 
75 1b e8 81 00 8a 36 13 00 fe ce 8b 0e 15 00 86 
cd c0 e1 06 0a 0e 11 00 31 c0 f8 eb 65 80 fc 02 
72 cb 80 fc 04 77 c6 60 80 cc 40 50 be 00 00 c7 
04 10 00 30 e4 89 44 02 89 5c 04 8c 44 06 66 31 
c0 66 89 44 0c 88 f0 f6 26 11 00 88 cf 88 eb c0 
ef 06 81 e1 3f 00 01 c8 48 89 c7 a1 13 00 f7 26 
11 00 f7 e3 01 f8 81 d2 00 00 89 44 08 89 54 0a 
58 30 c0 8a 16 10 00 e8 0c 00 88 26 03 00 61 a1 
02 00 1f ca 02 00 9c ff 1e 22 00 c3 80 fa 8f 7f 
04 88 16 2d 06 be 87 07 e8 8d 00 be be 07 31 c0 
b9 04 00 f6 04 80 74 03 40 89 f5 81 c6 10 00 e2 
f2 48 74 02 cd 18 bf 05 00 be 1d 06 c7 44 02 01 
00 66 8b 46 08 66 89 44 08 b8 00 42 8a 16 2d 06 
cd 13 73 0d 4f 74 49 30 e4 8a 16 2d 06 cd 13 eb 
d8 a1 fe 7d 3d 55 aa 75 37 fa 66 a1 4c 00 66 a3 
3f 06 be 13 04 8b 04 48 89 04 c1 e0 06 8e c0 31 
ff be 1d 06 b9 60 00 fc f3 a5 c7 06 4c 00 17 00 
a3 4e 00 fb 8a 16 2d 06 89 ee fa ea 00 7c 00 00 
be aa 07 e8 02 00 eb fe ac 20 c0 74 09 b4 0e bb 
07 00 cd 10 eb f2 c3 53 74 61 72 74 20 62 6f 6f 
74 69 6e 67 20 66 72 6f 6d 20 55 53 42 20 64 65 
76 69 63 65 2e 2e 2e 0d 0a 00 42 6f 6f 74 20 66 
61 69 6c 65 64 00 00 00 ea eb d4 ca 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 04 
05 00 0c fe ff ff 00 01 00 00 24 7a 94 03 55 aa 

```

sakumisu, 2023-01-18:
`Failed to enable port 1` means your power is not good or your usb circuit is not well.

arm000, 2023-01-18:
I also have some Ox64 boards, and I've confirmed it also fails the same way on both USB power and bench power.  I tried 2 units of each board type to rule out any defect specific to a particular board or board instance.

I did notice that occasionally I get this error message instead, but most of the time it's the "Failed to enable port 1" error:

[I/USB] New full-speed device on Hub 1, Port 1 connected
[E/USB] Failed to get device descriptor,errorcode:-116
[E/USB] Port 1 enumerate fail

Fishwaldo, 2023-01-18:
Is there a schematic of the devkit available? We could compare to see if any necessary components etc are missing from the ox64

sakumisu, 2023-01-18:
From the schematic, it is too simple in ox64 and M1s. Diff  90OHM and others, this need to contact pin64 and then contact us to check.

Fishwaldo, 2023-02-06:
@sakumisu any update on this? I noticed you deleted your comment about the circuit not being correct? Is the pine64 and m1s circuitry OK or is there a design issue?

grant-olson, 2023-02-09:
What I am seeing here. I have 3 devices, two fail and one works. If I plug them all in to a full PC, the two failing devices identify as USB 1.1 devices. The one that works is a 2.0 device. The device version can be found by running `lsusb -v` and looking at the `bcdUSB` field.

grant-olson, 2023-02-09:
Correction. The FAILING devices are 2.0. The SUCCESSFUL devices are 1.1.

grant-olson, 2023-02-09:
Apologies for so many messages. Actually I'm getting a combination of results. Its inconclusive.

closed 2023-05-31 by sakumisue