bouffalolab/bouffalo_sdk #67: BFLOG failes to compile on bl808

Fishwaldo, 2023-01-18:
Seems a linker section was missed from the bl808 mcu linker script. Attempting to compile the examples/bflog/barebone_sync failes with the following:
```[ 99%] Building C object CMakeFiles/barebone_sync_bl808_m0.elf.dir/main.c.obj
[100%] Linking C executable build_out/barebone_sync_bl808_m0.elf
/home/fish/bl808/bin/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: build_out/lib/libbflog.a(bflog.c.obj): in function `bflog_filter_set':
/home/fish/bl808/bl_mcu_sdk/components/bflog/bflog.c:505: undefined reference to `__bflog_tags_start__'
/home/fish/bl808/bin/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/fish/bl808/bl_mcu_sdk/components/bflog/bflog.c:505: undefined reference to `__bflog_tags_end__'
/home/fish/bl808/bin/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/fish/bl808/bl_mcu_sdk/components/bflog/bflog.c:505: undefined reference to `__bflog_tags_start__'
/home/fish/bl808/bin/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/fish/bl808/bl_mcu_sdk/components/bflog/bflog.c:505: undefined reference to `__bflog_tags_end__'
/home/fish/bl808/bin/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/fish/bl808/bl_mcu_sdk/components/bflog/bflog.c:502: undefined reference to `__bflog_tags_start__'
/home/fish/bl808/bin/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: build_out/lib/libbflog.a(bflog.c.obj): in function `bflog_flush':
/home/fish/bl808/bl_mcu_sdk/components/bflog/bflog.c:1187: undefined reference to `__bflog_tags_start__'
/home/fish/bl808/bin/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/fish/bl808/bl_mcu_sdk/components/bflog/bflog.c:1182: undefined reference to `__bflog_tags_end__'
/home/fish/bl808/bin/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/fish/bl808/bl_mcu_sdk/components/bflog/bflog.c:1182: undefined reference to `__bflog_tags_start__'
/home/fish/bl808/bin/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/fish/bl808/bl_mcu_sdk/components/bflog/bflog.c:1194: undefined reference to `__bflog_tags_end__'
/home/fish/bl808/bin/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: build_out/lib/libbflog.a(bflog.c.obj): in function `bflog':
/home/fish/bl808/bl_mcu_sdk/components/bflog/bflog.c:998: undefined reference to `__bflog_tags_start__'
/home/fish/bl808/bin/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/fish/bl808/bl_mcu_sdk/components/bflog/bflog.c:998: undefined reference to `__bflog_tags_start__'
/home/fish/bl808/bin/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/fish/bl808/bl_mcu_sdk/components/bflog/bflog.c:998: undefined reference to `__bflog_tags_end__'
/home/fish/bl808/bin/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/fish/bl808/bl_mcu_sdk/components/bflog/bflog.c:998: undefined reference to `__bflog_tags_end__'
```

sakumisu, 2023-01-18:
just ignore, please disable bflog

sakumisu, 2023-01-18:
Update code.