bouffalolab/bouffalo_sdk #69: Publish CMSIS-SVD files for each mcu, for tools to autogenerate code for register maps

TheButlah, 2023-01-18:
[CMSIS-SVD](https://www.keil.com/pack/doc/CMSIS/SVD/html/svd_Format_pg.html) files describe the register maps of the MCU, and are machine readable xml. There are automated tools, like [`svd2rust`](https://github.com/rust-embedded/svd2rust) that can take these files and generate code from them to allow languages to access the hardware peripherals. 

Lacking these files means that writing a library to access the peripherals from languages other than c is a highly tedious and error-prone process.

Could the SVD files for bouffalo's mcus be published?

sfranzyshen, 2023-01-20:
https://github.com/bouffalolab/bl808-pac/blob/main/bl808.svd

TheButlah, 2023-01-20:
I knew about the bl808 svd, but I was asking more about other chips too. Like for example, the BL616/BL618 and others

sfranzyshen, 2023-01-22:
not sure if this supports all the soc's available in the bl_mcu_sdk ...
https://github.com/openbouffalo/svd

TheButlah, 2023-01-22:
looks like the code parses the c headers from `bl_mcu_sdk` and outputs some sort of json file. I wonder what the next step is to go from that to an svd file

9names, 2023-01-23:
the obvious answer is you write an SVD generator. preferably less hacked together than mine, but it's good enough to generate an SVD that's no worse than the bl602 one without much effort.
https://github.com/9names/bl808-data/blob/main/bl808-data/src/bin/svd_generator.rs

of course I also wrote a header scraper to get the register data out of the SDK, since that's the only real source of info and the other scraping tool wasn't published yet.
Current plan is to dump out the data to separate files and manually build it back up. The SVD at bl808-pac is higher quality than the scraped data, and we'll want to improve it over time.

Current autogenerated SVD from my tool is at:
https://github.com/9names/bl808-pac/blob/alt/bl808.svd

brainstorm, 2023-02-26:
> I knew about the bl808 svd, but I was asking more about other chips too. Like for example, the BL616/BL618 and others

https://github.com/bouffalolab/bl-pac/blob/main/bl616/bl616.svd

closed 2023-02-26 by TheButlah