bouffalolab/bouffalo_sdk #71: Explaining UART clocks

mindstorm38, 2023-01-22:
Hi, I'm trying to understand how UART clock should be configured.

The reference manual clock diagram just shows one demux to select between `xclk`, `muxpll_160m_clk` and `mcu_bclk`:

[image]

The issue is that in the board support code for peripheral clock initialization, we first see a clock enable thing, which is not mentionned in the diagram (like all enable signals, but it's not a problem because it's straightforward):
https://github.com/bouffalolab/bl_mcu_sdk/blob/607c24d7fe4ae0cea6ddda7e660c6caccbf3e23d/bsp/board/bl808dk/board.c#L51

What I don't understand is the following code:
https://github.com/bouffalolab/bl_mcu_sdk/blob/607c24d7fe4ae0cea6ddda7e660c6caccbf3e23d/bsp/board/bl808dk/board.c#L63-L64

Why there are two demux selector registers and two divider ones (one in HBN and another in MM_GLB)? Also, there exist a `GLB_Set_DSP_UART1_CLK` function for UART1 but not for UART2, in this case how should we configure UART2?

sakumisu, 2023-01-22:
Only UART3 use `GLB_Set_DSP_UART0_CLK` ,uart0 ~2 use ` GLB_Set_UART_CLK(ENABLE, HBN_UART_CLK_XCLK, 0); 
`

mindstorm38, 2023-01-22:
So the naming is confusing? And what about `GLB_Set_DSP_UART1_CLK` ?

sakumisu, 2023-01-22:
Just ignore.

sakumisu, 2023-01-22:
UART3 is d0 first uart, named uart0 ,no confuse.

mindstorm38, 2023-01-22:
Ok, ty

closed 2023-01-22 by mindstorm38