bouffalolab/bouffalo_sdk #74: Error in PERIPHERAL_CLOCK_I2S_ENABLE() for BL70x.

pvvx, 2023-01-23:
[PERIPHERAL_CLOCK_I2S_ENABLE()](https://github.com/bouffalolab/bl_mcu_sdk/blob/master/drivers/lhal/include/bflb_clock.h#L147)

```
#define PERIPHERAL_CLOCK_I2S_ENABLE()                             \
    do {                                                          \
        volatile uint32_t regval = getreg32(BFLB_GLB_CGEN1_BASE); \
        regval |= (1 << 27);                                      \
        putreg32(regval, BFLB_GLB_CGEN1_BASE);                    \
    } while (0)
```
BL702/704/706 Reference Manual:
3.3.9 cgen_cfg1
Address：0x40000024 
i2s and qdec2 clock ungate enable - bit 26

sakumisu, 2023-01-24:
I2S lhal driver only works for bl616 and bl808 now.

pvvx, 2023-01-25:
I connected PCM1802 for tests and got it working on all CLK variants.

[image]

Stereo 96 kHz 24 bit is successfully transmitted from PCM1802 and BL702 via USB.
The PDS_Set_Audio_PLL_Freq() function has only 5 fixed frequencies, which is inconvenient.
For the project, I needed to change the MCLK frequency, but there is no documentation on setting up the AUDIO PLL.
I had to make research to generate any frequency from 0.5 to 80 MHz with a small step to output from the chip (GLB_Set_I2S_CLK(ENABLE, GLB_I2S_OUT_REF_CLK_SRC)) ;

sakumisu, 2023-01-27:
Thank you for such a detailed explanation. This will be completed later

closed 2023-01-27 by pvvx