bouffalolab/bouffalo_sdk #78: BL702 and GPIO24

Fishwaldo, 2023-01-27:
Hi,
I'm trying to bring up the [Sipeed M0Sense board](https://dl.sipeed.com/shareURL/Maix-Zero/Maix-Zero/2_Schematic), and doing a simple blinky test right now with the onboard led
It has a RGB led connected to GPIO23 (Blue) GPIO24 (green) and GPIO25 (Red). 

When enabling GPIO24 as  output with pullup, the board hangs shortly after setting the bflb_gpio_init:
```bflb_gpio_init(GPIO_PIN_24, GPIO_OUTPUT|GPIO_PULLUP|GPIO_DRV_1)```

(the other GPIO - 23, and 25 are fine)

Looking through the source, I can see some references to GPIO 24 in ```bflb_sf_cfg_init_ext_flash_gpio```,  which seems related to external flash, and thus not applicable to this board. Any suggestions to whats going on?

sakumisu, 2023-01-27:
gpio driver does not support  pin 24 ~ 28 init in BL702. It is special, so you can use `GLB_GPIO_Init` instead.

Fishwaldo, 2023-01-28:
Ok. I'll give it a try. 

By the way, I notice that almost none of the bflb_* functions in the lhal return any status (success/failure). You might want to consider that in a future major version so we can gracefully handle these cases?

sakumisu, 2023-01-28:
Some apis has return value, if not , it means it is not useful.

closed 2023-03-06 by sakumisue