bouffalolab/bouffalo_sdk #79: BL616 EMAC

ESP32DE, 2023-01-27:
Hi guys
新年快乐 2023! 

i play just in time with the BL616 and EMAC.
I noted from SDK the EMAC GPIO 

```
void board_emac_gpio_init()
{
    struct bflb_device_s *gpio;

    gpio = bflb_device_get_by_name("gpio");
    bflb_gpio_init(gpio, GPIO_PIN_25, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_26, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_27, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_28, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_29, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_30, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_31, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_32, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_33, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_34, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
#if defined(BL616)
    GLB_Config_AUDIO_PLL_To_400M();
    // GLB_PER_Clock_UnGate(GLB_AHB_CLOCK_IP_EMAC);
#endif
}

```

can we route missing pins

```

			BL616		BL618		BL616_SoC_PIN	BL617_SoC_PIN

RMII_REF_CLK				PAD_GPIO_25		-		40
RMII_TXD[0]				PAD_GPIO_26		-		41
RMII_TXD[1] 		PAD_GPIO_27	PAD_GPIO_27		31		43
RMII_RXD[0]		PAD_GPIO_28	PAD_GPIO_28		32		44
RMII_RXD[1]		PAD_GPIO_29	PAD_GPIO_29		33		45
RMII_RXERR		PAD_GPIO_30	PAD_GPIO_30		34		46
RMII_TX_EN				PAD_GPIO_31		-		47
RMII_RX_DV				PAD_GPIO_32		-		48
RMII_MDC				PAD_GPIO_33		-		49
RMII_MDIO 				PAD_GPIO_34		-		50

```

by simple using 

```
#elif defined(BL616)
#define GPIO_FUNC_SDH       (0 << GPIO_FUNC_SHIFT)
#define GPIO_FUNC_SPI0      (1 << GPIO_FUNC_SHIFT)
#define GPIO_FUNC_I2S       (3 << GPIO_FUNC_SHIFT)
#define GPIO_FUNC_PDM       (4 << GPIO_FUNC_SHIFT)
#define GPIO_FUNC_I2C0      (5 << GPIO_FUNC_SHIFT)
#define GPIO_FUNC_I2C1      (6 << GPIO_FUNC_SHIFT)
#define GPIO_FUNC_EMAC      (8 << GPIO_FUNC_SHIFT)                          // <------- use GPIO FUNC_8 -----)
#define GPIO_FUNC_CAM       (9 << GPIO_FUNC_SHIFT)
#define GPIO_FUNC_SDU       (12 << GPIO_FUNC_SHIFT)
#define GPIO_FUNC_PWM0      (16 << GPIO_FUNC_SHIFT)
#define GPIO_FUNC_AUDAC_PWM (25 << GPIO_FUNC_SHIFT)
#define GPIO_FUNC_JTAG      (26 << GPIO_FUNC_SHIFT)
#define GPIO_FUNC_PEC       (27 << GPIO_FUNC_SHIFT)
#define GPIO_FUNC_CLKOUT    (31 << GPIO_FUNC_SHIFT)

```

or did you this in the sdk and i did not found yet.

BTW: The "Hello World/Blinky"  [is working fine fast&clean out of the Box](https://twitter.com/eMbeddedHome/status/1618396061252620289)

thank you
best wishes
;-)

sakumisu, 2023-01-28:
I can not understand what do you mean? Using emac in bl616? If so ,sorry ,no enough pins to support emac, please use BL618.

ESP32DE, 2023-01-29:
> I can not understand what do you mean? Using emac in bl616? If so ,sorry ,no enough pins to support emac, please use BL618.

Hi

[Now I'm a little confused :-)  ](https://github.com/bouffalolab/bl_mcu_sdk/blob/e8da03a13baa050e94e58057aab4e3dc3b9a21f6/drivers/lhal/include/bflb_gpio.h#L24) 

[image]

from ... BL616 [should support ETH](https://github.com/bouffalolab/bl_mcu_sdk/blob/e8da03a13baa050e94e58057aab4e3dc3b9a21f6/drivers/soc/bl616/std/include/bl616_glb.h#L583-L586) - and here in[ board BL616dk](https://github.com/bouffalolab/bl_mcu_sdk/blob/e8da03a13baa050e94e58057aab4e3dc3b9a21f6/bsp/board/bl616dk/board.c#L327-L346) - EMAC not?

My suggestion would then be to route missing pins from free GPIO'S, 
Q: Can we do then like this

```
void board_emac_gpio_init()
{
    struct bflb_device_s *gpio;

    gpio = bflb_device_get_by_name("gpio");
 
// BL616 has IO25,IO26 
// should be ok not?
// routing must be done then to RMII_REF_CLK, RMII_TXD[0]
    bflb_gpio_init(gpio, GPIO_PIN_25, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_26, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
 
// BL616 has IO27,IO28,IO29,IO30 and is routed to 
// RMII_TXD[1], RMII_RXD[0], RMII_RXD[1], RMII_RXERR 
    bflb_gpio_init(gpio, GPIO_PIN_27, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_28, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_29, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_30, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);

#if defined(BL616) 
// BL616 only has IO's do 30 from Datasheet, so route other free IO's
// example IO11,IO12,IO13,IO14
// RMII_TX_EN, RMII_RX_DV, RMII_MDC, RMII_MDIO 
    bflb_gpio_init(gpio, GPIO_PIN_11, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_12, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_13, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_14, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1); 
#else    
    bflb_gpio_init(gpio, GPIO_PIN_31, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_32, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_33, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
    bflb_gpio_init(gpio, GPIO_PIN_34, GPIO_FUNC_EMAC | GPIO_ALTERNATE | GPIO_PULLUP | GPIO_SMT_EN | GPIO_DRV_1);
#endif
#if defined(BL616)
    GLB_Config_AUDIO_PLL_To_400M();
    // GLB_PER_Clock_UnGate(GLB_AHB_CLOCK_IP_EMAC);
#endif
}
```

txs

sakumisu, 2023-01-29:
BL616  does not have so many pins.Only 19 not 35.Board config is just a reference, if you want to use with other pins, modify by yourself.

ESP32DE, 2023-01-29:
> BL616 does not have so many pins.Only 19 not 35.Board config is just a reference, if you want to use with other pins, modify by yourself.

That answere is the right sound to my ears, buddy, 
so i  will do :) and try EMAC on BL616 :) i am working on it :)
please let it open, if possible, i will use LAN8720 btw
pcb is ready here :)

Txs

sakumisu, 2023-01-30:
We do not distinguish the same chip specially for drivers.

sakumisu, 2023-01-30:
The max number of pins is the standard.

closed 2023-02-26 by sakumisue

NickLiu635, 2024-10-31:
But Using RMII doesn't need to have more than 19 pins.