bouffalolab/bouffalo_sdk #80: BL702 usb device not working

andryblack, 2023-01-29:
usb cdc example freeze after request device descriptor

sakumisu, 2023-01-30:
Works good, no problem.

andryblack, 2023-01-30:
Crash is there: https://github.com/bouffalolab/bl_mcu_sdk/blob/master/drivers/lhal/src/bflb_usb_v1.c#L548

andryblack, 2023-01-30:
host is MacOS, presumably endpoint control data readed as out data.

sakumisu, 2023-01-30:
No other information i cannot help you.

gamelaster, 2023-01-30:
@andryblack it seems that macOS have some weird issue with USB CDC generally. I was forced to reduce CDC buffer sizes to get anything to work at all during ISP omde.

andryblack, 2023-01-30:
It is not cdc issue, crash occur on early, enumeration stage after request first descriptor. CDC on Mac OS works perfectly with correct code, for example with stm32 chips. I can produce minimal example.

LynnL4, 2023-04-12:
> @andryblack it seems that macOS have some weird issue with USB CDC generally. I was forced to reduce CDC buffer sizes to get anything to work at all during ISP omde.

I'm having the same problem on macos, can't upgrade via BLDevCude, how did you reduce the cdc buffer sizes to get the ISP to work, can you share? thanks

luyi1888, 2024-04-01:
I am experiencing the same issue.
SOC: BL702 Board:RV-Debugger Plus.
I think it may relevant to clock or host os.
I 'm less occur this issue on linux. 

However, I found the old sdk which sipeed used to develop RV-Debugger Plus are very stable.
https://github.com/sipeed/bl_mcu_sdk/tree/1cf3d1455b3baf66a9e3237365fcdfd69d821487
Demo project:
https://github.com/metro94/RV-DAP-Plus