bouffalolab/bouffalo_sdk #81: Does I2S Stereo in Slave mode have no channel synchronization?

pvvx, 2023-01-29:
When I2S is initialized in the DMA receive buffer, there is no certainty as to whether the receive data address is even or odd for the L or R channel. Depends on the I2S turn-on time. How to synchronize channels?
This is a hardware defect of all Bouffalo Lab chips?

pvvx, 2023-01-29:
Without a reset, the i2s will not re-sync.
```
static void reset_i2s0(void) {
	uint32_t tmpVal;
	tmpVal = getreg32(GLB_BASE+GLB_SWRST_CFG1_OFFSET);
	tmpVal |= GLB_SWRST_S1AA_MSK; // software reset I2S
	putreg32(tmpVal, GLB_BASE + GLB_SWRST_CFG1_OFFSET);
	tmpVal &= ~GLB_SWRST_S1AA_MSK; // software reset I2S
	putreg32(tmpVal, GLB_BASE + GLB_SWRST_CFG1_OFFSET);
}
```
After reset and initialization, you cannot stop DMA or clear the I2S FIFO - сhannel synchronization will be broken.

minlewang, 2023-01-31:
If your pcm data is 8 bits or 16 bits, you can use the channel merge feature to solve this problem

minlewang, 2023-01-31:
```
    /* enable channel merge */
    bflb_i2s_feature_control(i2s0, I2S_CMD_CHANNEL_LR_MERGE, true);
    /* enable i2s tx and rx */
    bflb_i2s_feature_control(i2s0, I2S_CMD_DATA_ENABLE, I2S_CMD_DATA_ENABLE_TX | I2S_CMD_DATA_ENABLE_RX);
```
You also need to change the bit width of the DMA's transmitted data.
This method can only work with 8/16bits of pcm data.

pvvx, 2023-01-31:
24 bits are used.
Solved by [workarounds with reset I2S](https://github.com/pvvx/PowerProfiler_pcm1802).
There are also situations in y I2C when reset is indispensable.
It is desirable to have macros or functions to reset controllers in the SDK.