bouffalolab/bouffalo_sdk #83: Add OSX riscv64 toolchain installation instructions

brainstorm, 2023-01-29:
See: https://github.com/riscv-collab/riscv-gnu-toolchain/issues/800#issuecomment-1312476123

Although it doesn't work out of the box (yet) since it needs some tweaking on the CMake files since some features of my M0S Dock are not working:

```
helloworld % make CHIP=bl616 BOARD=bl616dk
./../../tools/cmake/bin/cmake -S . -B build -G "Unix Makefiles" -DCROSS_COMPILE=riscv64-unknown-elf- -DCHIP=bl616 -DCPU_ID= -DBOARD=bl616dk -DBOARD_DIR= -DCONFIG_DEBUG=y -DCONFIG_ROMAPI=y -DCONFIG_USB_HS=y -DCONFIG_COMX=COM5 -DCMAKE_EXPORT_COMPILE_COMMANDS=OFF -DCONFIG_TLSF=y
-- [register library : libc], path:/Users/rvalls/dev/personal/bl_mcu_sdk/components/libc
-- [register library : mm], path:/Users/rvalls/dev/personal/bl_mcu_sdk/components/mm
-- [register library : utils], path:/Users/rvalls/dev/personal/bl_mcu_sdk/components/utils
-- [register library : lhal], path:/Users/rvalls/dev/personal/bl_mcu_sdk/drivers/lhal
-- [register extern library : libpka], path:/Users/rvalls/dev/personal/bl_mcu_sdk/drivers/lhal
-- [register library : std], path:/Users/rvalls/dev/personal/bl_mcu_sdk/drivers/soc/bl616/std
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/rvalls/dev/personal/bl_mcu_sdk/examples/helloworld/build
cd build && make -j8
Consolidate compiler generated dependencies of target app
Scanning dependencies of target std
[  2%] Building C object build_out/components/mm/CMakeFiles/mm.dir/mem.c.obj
[  2%] Building C object build_out/components/utils/CMakeFiles/utils.dir/bflb_block_pool/bflb_block_pool.c.obj
[  3%] Building C object build_out/components/utils/CMakeFiles/utils.dir/ring_buffer/ring_buffer.c.obj
[  4%] Building ASM object build_out/drivers/soc/bl616/std/CMakeFiles/std.dir/startup/start.S.obj
[  5%] Building C object build_out/components/utils/CMakeFiles/utils.dir/log/log.c.obj
[  5%] Building C object CMakeFiles/app.dir/Users/rvalls/dev/personal/bl_mcu_sdk/bsp/board/bl616dk/board.c.obj
[  6%] Building C object build_out/components/libc/CMakeFiles/libc.dir/nuttx/libc/stdlib/lib_abs.c.obj
[  7%] Building C object build_out/drivers/lhal/CMakeFiles/lhal.dir/src/bflb_common.c.obj
cc1: error: unknown cpu 'e907' for '-mtune'
cc1: error: unknown cpu 'e907' for '-mtune'
cc1: error: unknown cpu 'e907' for '-mtune'
cc1: error: unknown cpu 'e907' for '-mtune'
cc1: error: unknown cpu 'e907' for '-mtune'
cc1: error: unknown cpu 'e907' for '-mtune'
cc1: error: unknown cpu 'e907' for '-mtune'
cc1: error: unknown cpu 'e907' for '-mtune'
make[3]: *** [CMakeFiles/app.dir/Users/rvalls/dev/personal/bl_mcu_sdk/bsp/board/bl616dk/board.c.obj] Error 1
make[3]: *** [build_out/components/utils/CMakeFiles/utils.dir/bflb_block_pool/bflb_block_pool.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: *** [build_out/components/mm/CMakeFiles/mm.dir/mem.c.obj] Error 1
make[3]: *** [build_out/components/utils/CMakeFiles/utils.dir/log/log.c.obj] Error 1make[3]: *** [build_out/components/libc/CMakeFiles/libc.dir/nuttx/libc/stdlib/lib_abs.c.obj] Error 1
make[2]: *** [CMakeFiles/app.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [build_out/components/mm/CMakeFiles/mm.dir/all] Error 2

make[3]: *** [build_out/drivers/lhal/CMakeFiles/lhal.dir/src/bflb_common.c.obj] Error 1
make[3]: *** [build_out/components/utils/CMakeFiles/utils.dir/ring_buffer/ring_buffer.c.obj] Error 1
make[2]: *** [build_out/drivers/lhal/CMakeFiles/lhal.dir/all] Error 2
make[2]: *** [build_out/components/utils/CMakeFiles/utils.dir/all] Error 2
make[3]: *** [build_out/drivers/soc/bl616/std/CMakeFiles/std.dir/startup/start.S.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [build_out/components/libc/CMakeFiles/libc.dir/all] Error 2
[  8%] Building ASM object build_out/drivers/soc/bl616/std/CMakeFiles/std.dir/startup/vector.S.obj
cc1: error: unknown cpu 'e907' for '-mtune'
make[3]: *** [build_out/drivers/soc/bl616/std/CMakeFiles/std.dir/startup/vector.S.obj] Error 1
make[2]: *** [build_out/drivers/soc/bl616/std/CMakeFiles/std.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [build] Error 2
```

sakumisu, 2023-01-29:
Must use our toolchain in readme

sakumisu, 2023-01-29:
Your toolchain is not match with our chip.

gamelaster, 2023-01-29:
Follow this: https://github.com/p4ddy1/pine_ox64/blob/main/build_toolchain_macos.md

brainstorm, 2023-02-18:
> Follow this: https://github.com/p4ddy1/pine_ox64/blob/main/build_toolchain_macos.md

I followed those instructions and for a repo cloned today, the compilation stalls here (or whatever the command after `echo timestamp > s-output` actually is):

```
Statistics for recog:
  Number of decisions: 257040
  longest path:           193 (code:    921)
  longest backtrack:       15 (code:  60790)
Statistics for split_insns:
  Number of decisions:    834
  longest path:            51 (code:     46)
  longest backtrack:        5 (code:     41)
Statistics for peephole2_insns:
  Number of decisions:     90
  longest path:            22 (code:     13)
  longest backtrack:        1 (code:      9)
Shared 187991 out of 392069 states by creating 8887 new states, saving 179104
/bin/sh ../.././riscv-gcc/gcc/../move-if-change tmp-recog.c insn-recog.c
recogv=$(echo $(csplit insn-recog.c /parallel\ compilation/ -k -s {20} -f insn-recog -b "%d.c" 2>&1));\
	[ ! "$recogv" ] || grep "match not found" <<< $recogv
csplit: ‘/parallel compilation/’: match not found
[ -s insn-recog0.c ] || (for i in 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21; do touch insn-recog$i.c; done && echo "" > insn-recog.c)
echo timestamp > s-recog
/bin/sh ../.././riscv-gcc/gcc/../move-if-change tmp-automata.c insn-automata.c
automatav=$(echo $(csplit insn-automata.c /parallel\ compilation/ -k -s {20} -f insn-automata -b "%d.c" 2>&1));\
	[ ! "$automatav" ] || grep "match not found" <<< $automatav
csplit: ‘/parallel compilation/’: match not found
[ -s insn-automata0.c ] || (for i in 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21; do touch insn-automata$i.c; done && echo "" > insn-automata.c)
echo timestamp > s-automata
/bin/sh ../.././riscv-gcc/gcc/../move-if-change tmp-attrtab.c    insn-attrtab.c
/bin/sh ../.././riscv-gcc/gcc/../move-if-change tmp-dfatab.c     insn-dfatab.c
/bin/sh ../.././riscv-gcc/gcc/../move-if-change tmp-latencytab.c insn-latencytab.c
echo timestamp > s-attrtab
/bin/sh ../.././riscv-gcc/gcc/../move-if-change tmp-output.c insn-output.c
outputv=$(echo $(csplit insn-output.c /parallel\ compilation/ -k -s {20} -f insn-output -b "%d.c" 2>&1));\
	[ ! "$outputv" ] || grep "match not found" <<< $outputv
csplit: ‘/parallel compilation/’: match not found
[ -s insn-output0.c ] || (for i in 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21; do touch insn-output$i.c; done && echo "" > insn-output.c)
echo timestamp > s-output
```

brainstorm, 2023-02-18:
Compiled the toolchain and the `helloworld` example successfully... BUT:

```
% make flash CHIP=bl616 COMX=/dev/tty.usbmodem1102
Support in next version, please use devcube now
```

So what's devcube exactly? A hardware probe? A windows program? Is there something equivalent for OSX?

Fishwaldo, 2023-02-18:
You have to flash with BLDevCube

closed 2023-02-18 by brainstorm