bouffalolab/bouffalo_sdk #85: refactoring ETA?

sfranzyshen, 2023-01-31:
There are several issues here that are answered with **"refactoring"**

Is There any ETA available on when this **"refactoring"** will be done and available?
will it be one large update ... or will there be smaller updates spread out over several?

sakumisu, 2023-02-01:
Just code style update.

sfranzyshen, 2023-02-01:
> Just code style update.

can you give us some kind of time frame? more than 1 Month or 1 Year?

sakumisu, 2023-02-01:
Not sure, most of drivers are done, wireless is only released for customers.

pzeiler, 2023-02-01:
> Not sure, most of peripheral drivers are done like lhal, wireless is only released for customers.

Could you please clarify what is meant by "wireless is only released for customers"? There will be no ZigBee/BLE/WiFi support publicly available?

pzeiler, 2023-02-02:
Having an open source RISC-V CPU and even closed source drivers for it's periphals sounds like a really ugly idea. But having no driver for wireless peripherals at all would be outrageous. I am also a customer, maybe I only bought a handful of boards, but I am still a customer. 
Why do you think the ESP/ESP32 platform was such a huge success? Certainly not because they limited the possibilites of the makers praising them highly. 
I also don't get how Pine64 (which should be a "real" customer by your standards) would accepted a CPU with a limited SDK for their Ox64-Board, clearly aimed at the makerspace. And it is even marketed with all the nice wireless bells & whistles...

Iscle, 2023-02-02:
I agree with @pzeiler, this board chip would be a huge success if it wasn't for the poor SDK support and the lack of wireless functionality (software-wise).

The chip itself is amazing, multi-core, with high-speed usb, wifi, bluetooth, zigbee, h264 encoder, mmu... It's a dream for many of us and I was very excited when I bought it, but now I'm already considering other options because of your attitude towards the OSS community.

sakumisu, 2023-02-02:
You can familiarize yourself with the peripherals (no wireless)first.

gamelaster, 2023-02-02:
@Iscle and @pzeiler The BL808 WiFi driver is already in bl_iot_sdk bl808 branch, and in SiPeed BL808 SDK.
(note, my opinion) Thus, I think what Bouffalo is trying to say, is that Wireless will be added at some point, as it is not fully ready for public release.

Iscle, 2023-02-02:
> You can familiarize yourself with the peripherals (no wireless)first.

We know it's hard to develop an entire SDK for a brand new chip, and we greatly appreciate the work and effort you're putting into it. 
We're not requesting a 100% working, bug-free SDK for tomorrow, we're only requesting an estimate, in order for us to be able to adapt our workflow, consider other options and organise or projects based on that.

For example, my project requires WiFi + Bluetooth Classic. + USB to work. I already have the USB part working, for WiFi I can get the closed-source lib from Sipeed SDK (that's not an issue), but there's no estimate, library or even mention about Bluetooth Classic / BLE anywhere.

I hope you understand our point :)

Thanks again for the work and contributing to OSS.

sfranzyshen, 2023-02-02:
by the time this company gets their crap together ... Espressif or raspberrypi will have a cheaper equivalent ... and bouffalolab will still be waiting on the NDA ...
(note, my opinion)

ESP32DE, 2023-02-03:
I'm just plugging in here to stay up to date.

sakumisu, 2023-02-04:
Thank you for your support and anticipation. This part of wireless open source is time and labour intensive, so please be patient.
SDK2.0 will support both mcu and iot, so you can familiarise yourself with the mcu part of the peripheral first and give us feedback on any problems you may have, so that you can use it soon after the wireless support is up.

BTW, you can get wifi from other sdk like sipeed sdk or iot sdk, if you want to use in SDK2.0, please wait. And i am not sure the release time.

pzeiler, 2023-02-04:
Thank you for your reply. Could you also give any information about ZigBee availability in the SDK2.0? A ZNP approach would be fine for my project...

gamelaster, 2023-02-04:
ZNP approach?

pzeiler, 2023-02-04:
> ZNP approach?

Like not handling it on the low level, just init the service and let the service handle things like network discovery, security, service advertisement etc.

pzeiler, 2023-02-04:
> 

https://software-dl.ti.com/simplelink/esd/simplelink_cc26x2_sdk/2.30.00.34/exports/docs/zstack/html/zigbee/znp_interface.html

angelovAlex, 2023-03-06:
time is ticking... we are still waiting... new chips are being released...

sakumisue, 2023-03-31:
done

closed 2023-03-31 by sakumisue