bouffalolab/bouffalo_sdk #89: netif static IP address seems to be ignored

alexhorner, 2023-02-17:
I have tried a couple of example programs based on netif now, specifically the `lwip_tcp` and `lwip_udp` examples.

When updating `IP_ADDR0` - `IP_ADDR3` the new address gets ignored, and the old address `192.168.123.100` is used instead.

I have cleaned and rebuilt, and reflashed, but it is ignored again.

I am on commit `18408f971e3f8c2f82e79ec5fddd38c22f288c0d`

sakumisu, 2023-02-18:
See in `ethernet_link_status_updated`

alexhorner, 2023-02-18:
> See in `ethernet_link_status_updated`

I am not sure what to look for? It just prints the IP that is set, but the set IP is incorrect, meaning it prints the incorrect IP address.

I have set the IP like so

```c
/*Static IP ADDRESS: IP_ADDR0.IP_ADDR1.IP_ADDR2.IP_ADDR3 */
#define IP_ADDR0 (uint8_t)10
#define IP_ADDR1 (uint8_t)0
#define IP_ADDR2 (uint8_t)10
#define IP_ADDR3 (uint8_t)197

// Local NET Mask address define
/*NETMASK*/
#define NETMASK_ADDR0 (uint8_t)255
#define NETMASK_ADDR1 (uint8_t)255
#define NETMASK_ADDR2 (uint8_t)255
#define NETMASK_ADDR3 (uint8_t)0

// Local Gateway address define
/*Gateway Address*/
#define GW_ADDR0 (uint8_t)10
#define GW_ADDR1 (uint8_t)0
#define GW_ADDR2 (uint8_t)10
#define GW_ADDR3 (uint8_t)1
```

Then later it does

```c
/**
  * @brief  Setup the network interface
  * @param  None
  * @retval None
  */
static void netif_config(void)
{
    ip_addr_t ipaddr;
    ip_addr_t netmask;
    ip_addr_t gw;

/* if use DHCP */
#if LWIP_DHCP
    ip_addr_set_zero_ip4(&ipaddr);
    ip_addr_set_zero_ip4(&netmask);
    ip_addr_set_zero_ip4(&gw);
#else

    /* IP address default setting */
    IP4_ADDR(&ipaddr, IP_ADDR0, IP_ADDR1, IP_ADDR2, IP_ADDR3);
    IP4_ADDR(&netmask, NETMASK_ADDR0, NETMASK_ADDR1, NETMASK_ADDR2, NETMASK_ADDR3);
    IP4_ADDR(&gw, GW_ADDR0, GW_ADDR1, GW_ADDR2, GW_ADDR3);

#endif

    /* add the network interface */
    netif_add(&gnetif, &ipaddr, &netmask, &gw, NULL, &ethernetif_init, &tcpip_input);

    /*  Registers the default network interface */
    netif_set_default(&gnetif);
    ethernet_link_status_updated(&gnetif);

#if LWIP_NETIF_LINK_CALLBACK
    netif_set_link_callback(&gnetif, ethernet_link_status_updated);
#endif
}
```

But the output is:

```

  ____               __  __      _       _       _
 |  _ \             / _|/ _|    | |     | |     | |
 | |_) | ___  _   _| |_| |_ __ _| | ___ | | __ _| |__
 |  _ < / _ \| | | |  _|  _/ _` | |/ _ \| |/ _` | '_ \
 | |_) | (_) | |_| | | | || (_| | | (_) | | (_| | |_) |
 |____/ \___/ \__,_|_| |_| \__,_|_|\___/|_|\__,_|_.__/

Build:22:17:37,Feb 17 2023
Copyright (c) 2022 Bouffalolab team
=========== flash cfg ==============
jedec id   0xEF4018
mid            0xEF
iomode         0x04
clk delay      0x01
clk invert     0x01
read reg cmd0  0x05
read reg cmd1  0x35
write reg cmd0 0x01
write reg cmd1 0x31
qe write len   0x01
cread support  0x00
cread code     0xFF
burst wrap cmd 0x77
=====================================
dynamic memory init success,heap size = 125 Kbyte
sig1:ffff32ff
sig2:0000ffff
EMAC lwip test case !
lwip statck init
lwip init done
sys_mbox_new done!
sys_mutex_new done!
tcpip thread init done!
netif config
emac init
emac phy addr:0x0001
emac phy reg read 0x60e1
emac phy id 1 =00000007
emac phy id 2 =0000c0f1
ETH PHY init ok!
PHY[7c0f1] @1 ready on 100Mbps, full duplex
EMAC start
[OS] Starting emac rx task...
[OS] bl Netif is up
Static IP address: 192.168.123.100
[OS] Starting tcp server task...
TCP Server create socket
Create server success, server ip & listen port:192.168.123.100:3365
```

I have not specified `192.168.123.100` anywhere in this built file.

sakumisu, 2023-02-18:
Don't you see IP config in `ethernet_link_status_updated` ?

sakumisu, 2023-02-18:
Must modify macros in ethernet.c, others are not useful.

sakumisu, 2023-02-18:
https://github.com/bouffalolab/bl_mcu_sdk/blob/master/bsp/common/ethernet/ethernetif.c#L58

alexhorner, 2023-02-18:
Found it, thank you. What is the purpose of the other one if it cannot be used?

sakumisu, 2023-02-19:
Just personal writing habits, you can modify by yourself.

closed 2023-02-20 by alexhorner