bouffalolab/bouffalo_sdk #93: Note: Flash with new command tool or devcube>=1.8.2

sakumisu, 2023-02-24:
If you use new flash tool, it adds bootheader in firmware, so if you use devcube to download firmware, must use devcube version >= 1.8.2

cjacker, 2023-02-27:
After BLFlashCommand added and the commit [[update][board] enable fw header for new flash tool ](https://github.com/bouffalolab/bl_mcu_sdk/commit/e70e482d2129411f34208d1184b4710074c67777). it result in:

- flashing with blisp and bflb-mcu-tool completely broken. it also broken all tutorials, howtos, etc...
- BLFlashCommand not works as expected, the error as below, report missing "eflash_loader_32m.bin".
- There is no source codes released for BLFlashCommand and various eflash_loader.elf, etc...

**My Questions:**
- Is it really good to commit so many binaries and break all known open-source solutions?
- Is 'bflb-mcu-tool' also official program tool? How is it different from 'BLFlashCommand'? which is maintained and supported by official in deed?

BLFlashCommand error output:
```
./BLFlashCommand --interface uart --port /dev/ttyACM0 --chipname bl702 --config ./flash_prog_cfg.ini
['./BLFlashCommand', '--interface', 'uart', '--port', '/dev/ttyACM0', '--chipname', 'bl702', '--config', './flash_prog_cfg.ini']
[10:27:10.357] - Serial port is /dev/ttyACM0
[10:27:10.357] - ==================================================
[10:27:10.359] - Program Start
[10:27:10.359] - ========= eflash loader cmd arguments =========
[10:27:10.360] - serial port is /dev/ttyACM0
[10:27:10.360] - chiptype: bl702
[10:27:10.360] - cpu_reset=False
[10:27:10.377] - ========= Interface is uart =========
[10:27:10.377] - Eflash load bin file: /home/cjacker/bouffalo_flash_cube/chips/bl702/eflash_loader/eflash_loader_32m.bin
[10:27:10.377] - ========= load eflash_loader.bin =========
[10:27:10.377] - Load eflash_loader.bin via uart
[10:27:10.377] - ========= image load =========
[10:27:10.389] - default set DTR high
[10:27:10.490] - usb serial port
[10:27:10.541] - clean buf
[10:27:10.542] - send sync
[10:27:10.753] - ack is b'4f4b464c0201'
[10:27:10.784] - shake hand success
[10:27:10.794] - get_boot_info
[10:27:10.798] - data read is b'0100020700000000931d000027000004000043575c4cb97c'
[10:27:10.798] - ========= ChipID: 000043575c4cb97c =========
[10:27:10.798] - last boot info: None
[10:27:10.798] - sign is 0 encrypt is 0
[10:27:10.798] - [Errno 2] No such file or directory: '/home/cjacker/bouffalo_flash_cube/chips/bl702/eflash_loader/eflash_loader_32m.bin'
Traceback (most recent call last):
  File "libs/base/bflb_img_loader.py", line 941, in img_load_process
  File "libs/base/bflb_img_loader.py", line 802, in img_load_main_process
FileNotFoundError: [Errno 2] No such file or directory: '/home/cjacker/bouffalo_flash_cube/chips/bl702/eflash_loader/eflash_loader_32m.bin'
[10:27:10.800] - Load helper bin time cost(ms): 422.57763671875
[10:27:10.800] - {"ErrorCode": "0003","ErrorMsg":"BFLB LOAD HELP BIN FAIL"}
[10:27:10.802] - Burn Retry
[10:27:10.802] - Burn return with retry fail
```

sakumisu, 2023-02-27:
ALL are officiial. If bflb-mcu-tool removed bootheader and other functions, it is BLFlashCommand

sakumisu, 2023-02-27:
bin has ignored by .gitignore.

sakumisu, 2023-02-27:
I hate sync from internal to github, we use repo tool.

cjacker, 2023-02-27:
Simple questions:

I build a simple led_blink project as usual with git master, the result is 'sipeed_rv_debugger_plus_blink.bin'.

- how can I process this bin file to program it with 'blisp' or 'bflb-mcu-tool' ? it can programmed as before, but not works.
- how can I program it with 'BLFlashCommand'? the detail usage needed.

sakumisu, 2023-02-27:
please see doc.

Fishwaldo, 2023-02-27:
Can I suggest you setup GitHub actions to build all the examples with different Chips and then sync from internal via pull requests instead.  Then you can catch build errors before they break things for everyone?

Also, can you start tagging releases so we can pick stable commits?

sakumisu, 2023-02-27:
`sipeed_rv_debugger_plus_blink.bin` add bootheader then you can use new flash tool.

cjacker, 2023-02-27:
It not works at all, as I mentioned above, the 'eflash_loader_32m.bin' is missing, where can I find it?

```
$ make flash COMX=/dev/ttyACM0
./../../tools/bflb_tools/bouffalo_flash_cube/BLFlashCommand \
--interface=uart \
--baudrate=2000000 \
--port=/dev/ttyACM0 \
--chipname=bl702 \
--cpu_id= \
--config=flash_prog_cfg.ini
['./../../tools/bflb_tools/bouffalo_flash_cube/BLFlashCommand', '--interface=uart', '--baudrate=2000000', '--port=/dev/ttyACM0', '--chipname=bl702', '--cpu_id=', '--config=flash_prog_cfg.ini']
[11:01:17.957] - Serial port is /dev/ttyACM0
[11:01:17.958] - ==================================================
[11:01:17.960] - Program Start
[11:01:17.960] - ========= eflash loader cmd arguments =========
[11:01:17.961] - serial port is /dev/ttyACM0
[11:01:17.961] - chiptype: bl702
[11:01:17.962] - cpu_reset=False
[11:01:17.980] - ========= Interface is uart =========
[11:01:17.980] - Eflash load bin file: /home/cjacker/bl_mcu_sdk/tools/bflb_tools/bouffalo_flash_cube/chips/bl702/eflash_loader/eflash_loader_32m.bin
[11:01:17.980] - ========= load eflash_loader.bin =========
[11:01:17.980] - Load eflash_loader.bin via uart
[11:01:17.980] - ========= image load =========
[11:01:17.993] - default set DTR high
[11:01:18.093] - usb serial port
[11:01:18.144] - clean buf
[11:01:18.145] - send sync
[11:01:18.356] - ack is b'4f4b464c0201'
[11:01:18.386] - shake hand success
[11:01:18.397] - get_boot_info
[11:01:18.399] - data read is b'0100020700000000931d000027000004000043575c4cb97c'
[11:01:18.400] - ========= ChipID: 000043575c4cb97c =========
[11:01:18.400] - last boot info: None
[11:01:18.400] - sign is 0 encrypt is 0
[11:01:18.400] - [Errno 2] No such file or directory: '/home/cjacker/bl_mcu_sdk/tools/bflb_tools/bouffalo_flash_cube/chips/bl702/eflash_loader/eflash_loader_32m.bin'
Traceback (most recent call last):
  File "libs/base/bflb_img_loader.py", line 941, in img_load_process
  File "libs/base/bflb_img_loader.py", line 802, in img_load_main_process
FileNotFoundError: [Errno 2] No such file or directory: '/home/cjacker/bl_mcu_sdk/tools/bflb_tools/bouffalo_flash_cube/chips/bl702/eflash_loader/eflash_loader_32m.bin'
[11:01:18.400] - Load helper bin time cost(ms): 419.472900390625
[11:01:18.400] - {"ErrorCode": "0003","ErrorMsg":"BFLB LOAD HELP BIN FAIL"}
[11:01:18.402] - Burn Retry
[11:01:18.402] - Burn return with retry fail
```

sakumisu, 2023-02-27:
I have said it is ignored by gitignore.

sakumisu, 2023-02-27:
Please update commit.

cjacker, 2023-02-27:
> Please update commit.

OK. Thanks for your on-line support.

I noticed you just commit in all bin files. after "git pull" again, it works now.

Please consider to provide 'BLFlashCommand' and bootloaders' source codes when your time is avaiblable.

sakumisu, 2023-02-27:
old firmware + bootheader = new firmware.

sakumisu, 2023-02-27:
Flash tool involved security, so we cannot release source code, sorry.

sakumisu, 2023-02-27:
> Can I suggest you setup GitHub actions to build all the examples with different Chips and then sync from internal via pull requests instead. Then you can catch build errors before they break things for everyone?
> 
> Also, can you start tagging releases so we can pick stable commits?

cjacker, 2023-02-27:
OK, finally I figured out how to deal with new firmware with blisp and bflb-mcu-tool.

new firmware have a section '.fw_header' added to final elf and from the building process output:

```
Image Offset:0x1000
```
The '.fw_header' section could be removed by objcopy? but I did not try.

using `blisp` with '-l 0x1000' and blfb-mcu-tool with '--addr 0x1000', these tool can program new firmware correctly and works.

sakumisu, 2023-02-27:
> OK, finally I figured out how to deal with new firmware with blisp and bflb-mcu-tool.
>
> new firmware have a section '.fw_header' added to final elf and from the building process output:
>
> ```
> Image [...]

because these tool use whole bin, so the same with new flash tool.

gamelaster, 2023-02-27:
@cjacker blisp should accept those images with boot header via IOT command, although, this will probably change in future, as blisp needs some refactoring together with CLI commands API change, to be able to properly address requirements of multi-core chips (BL808)

cjacker, 2023-02-27:
> @cjacker blisp should accept those images with boot header via IOT command, although, this will probably change in future, as blisp needs some refactoring together with CLI commands API change, to be able to properly address requirements of multi-core chips (BL808)

Thanks, I will try it later to verify.

cjacker, 2023-03-07:
> 

The aim of BLFlashCommand is good since it integrated with bl_mcu_sdk and 'make flash' works very well.

But a question here,  do we really need these different flash tools?

Here are some suggestions :

- bflb-iot-tool and bflb-mcu-tool can merged together and renamed to 'bl-mcu-tool' if it will provide more command line args.
- The IOT page and MCU page can also be merged together. there are not much difference at all.
- BLFlashCommand / GUI can be replaced by this new 'bl-mcu-tool' if it have the ability to read flash config file.

After that, we have a most opensource CLI tool and a pyinstaller packaged close source GUI tool. both have same features and can be replaced each other.

It eliminated all confusions, save a lot of time to maintain different projects, save a lot of time to  educate users, save a lot of time to write too many documents, and everybody can improve the CLI tool if they want.

Please consider it.

sakumisu, 2023-03-07:
No simple, sorry.

sakumisu, 2023-03-07:
Use our mcu sdk please use new flash tool, thanks.

cjacker, 2023-03-07:
> Use our mcu sdk please use new flash tool, thanks.

Today's update occupied about 70M+ disk space since all binaries bflb tools updated.

So request again, please consider open part of the source codes of BLFlashCommand, at least send a release to pypi some day just like bflb-mcu-tool and iot-tool.

And I really suggest you can read this tutorial I wrote recently when your time is available , and maybe you will be able to understand why so many choices make us confused from the users/customers/developers perspective.

https://github.com/cjacker/opensource-toolchain-bouffalo-lab

closed 2023-03-31 by sakumisue