bouffalolab/bouffalo_sdk #96: How to reset CPU via JTAG

gamelaster, 2023-02-24:
BL808: I want to flash firmware via JTAG, but I need to go to ISP mode. How to do this via CK-Link?
reset, sreset or nreset dones't work.

JF002, 2023-02-25:
I had the same issue with the BL618 and found the answer in the `.launch` file (config file for Eclipse, I think) in bl_mcu_sdk. According to [CKLink_BL808.launch](https://github.com/bouffalolab/bl_mcu_sdk/blob/master/CKLink_BL808.launch), you can reset the BL808 by issuing the following commands over GDB:

```
set $pc=0x58000000
set $mie=0
set $mstatus=0x1880

thb main
```

To make things easier, I defined a function and a hook in `.gdbinit`:
```
define bl-reset
set $pc=0xA0000000
set $mie=0
set $mstatus=0x1880
thb main
end

define target hookpost-remote
bl-reset
end
```

`bl-reset` allows you to manually reset the MCU, and the hook automatically resets it when the debugger connects to the board.
If I remember correctly, it was not perfect, but it's the best I could do.

I put all my notes together [in this blog post](https://codingfield.com/blog/2023-01/getting-started-with-bl618dk/#debugging-with-clion). Let me know if it works for you on the BL808!

sakumisu, 2023-02-25:
As @JF002 said, only this way, risc-v do not have register for hardware reset, and cklink do not support nrst and rst.

gamelaster, 2023-02-25:
@sakumisu yes, I know about this method, but the goal is to get into BootROM (0x90000000), so it is possible to flash the firmware via JTAG, without reseting board with buttons.
`set $pc=0x90000000` doesn't work, probably because of PMP?
`GLB_SW_System_Reset` reset's CPU, but also JTAG core :frowning_face:.

sakumisu, 2023-02-25:
please in boot mode then you can use ` set $pc=0x90000000`

madushan1000, 2023-02-25:
You can get it to halt in the bootrom by following this (https://gist.github.com/madushan1000/bb985be6e9f4b7c69ee16216f0060ece). I used pine64 jtag adapter(ft232hl chip). I think rv debugger plus will work too. Didn't check.