bouffalolab/bouffalo_sdk #97: ISO11898 example: Which pins for bl616's M0S Dock?

brainstorm, 2023-02-26:
Going through the CAN (ISO11898) example, it seems like the pins init function, [`board_iso11898_gpio_init()`](https://github.com/bouffalolab/bl_mcu_sdk/blob/c75ba0cf8403802a30912e89d2105969b4b73913/examples/peripherals/iso11898/iso11898_selftest/main.c#L43), is commented out [in the SDK itself](https://github.com/bouffalolab/bl_mcu_sdk/blob/c75ba0cf8403802a30912e89d2105969b4b73913/bsp/board/bl616dk/board.c#L464-L469) and therefore no pins are assigned anywhere?

brainstorm, 2023-03-04:
@gamelaster Following up over here from our Telegram discussion... you meant something like this?:

```diff
diff --git a/drivers/lhal/include/bflb_gpio.h b/drivers/lhal/include/bflb_gpio.h
index e4f410d8..1417c4b0 100644
--- a/drivers/lhal/include/bflb_gpio.h
+++ b/drivers/lhal/include/bflb_gpio.h
@@ -123,6 +123,7 @@
 #define GPIO_FUNC_AUDAC_PWM (25 << GPIO_FUNC_SHIFT)
 #define GPIO_FUNC_JTAG      (26 << GPIO_FUNC_SHIFT)
 #define GPIO_FUNC_PEC       (27 << GPIO_FUNC_SHIFT)
+#define GPIO_FUNC_ISO11898  (28 << GPIO_FUNC_SHIFT)
 #define GPIO_FUNC_CLKOUT    (31 << GPIO_FUNC_SHIFT)
 #elif defined(BL606P) || defined(BL808)
 #define GPIO_FUNC_SDH     (0 << GPIO_FUNC_SHIFT)
diff --git a/examples/peripherals/iso11898/iso11898_selftest/main.c b/examples/peripherals/iso11898/iso11898_selftest/main.c
index 7298d861..d2b6318a 100644
--- a/examples/peripherals/iso11898/iso11898_selftest/main.c
+++ b/examples/peripherals/iso11898/iso11898_selftest/main.c
@@ -1,4 +1,5 @@
 #include "bflb_mtimer.h"
+#include "bflb_gpio.h"
 #include "bflb_iso11898.h"
 #include "board.h"
 
@@ -40,9 +41,11 @@ int main(void)
 {
     int ret = 0;
     board_init();
-    board_iso11898_gpio_init();
+    //board_iso11898_gpio_init();
 
     iso11898 = bflb_device_get_by_name("iso11898");
+    bflb_gpio_init(iso11898, GPIO_PIN_16, GPIO_FUNC_ISO11898 | GPIO_ALTERNATE );
+    bflb_gpio_init(iso11898, GPIO_PIN_17, GPIO_FUNC_ISO11898 | GPIO_ALTERNATE );
 
     bflb_iso11898_init(iso11898, &cfg);
     bflb_iso11898_set_filter(iso11898, &filter);
```

Can't find in the datasheet where GPIOs can be mapped on the ISO11898 `&config` struct...

gamelaster, 2023-03-04:
@brainstorm this seems to be right.

brainstorm, 2023-03-04:
> @brainstorm this seems to be right.

No, it's not :/

brainstorm, 2023-03-04:
After looking at the current SDK function state, as mentioned in the first comment:

https://github.com/bouffalolab/bl_mcu_sdk/blob/c75ba0cf8403802a30912e89d2105969b4b73913/bsp/board/bl616dk/board.c#L464-L469

I believe that the CAN (ISO11898 peripheral) has not seen much testing other than selftest (no actual GPIO pins output), am I right, @sakumisue?

sakumisu, 2023-03-06:
> After looking at the current SDK function state, as mentioned in the first comment: [...]

No,just not updated.

sakumisu, 2023-03-06:
Updated now.

brainstorm, 2023-03-06:
I think that you closed this issue a bit too fast? I noticed your changes on [this commit lines](https://github.com/bouffalolab/bl_mcu_sdk/commit/eb7afedb206b2c41d2658da8f9c6a9027eb67b23#diff-f6366c338082d7378bc7f5ca0a68f09fb8534abef4ec9b9b2aec213400ceecaeR466-R470)

But when I plug my oscilloscope to GPIO 14 and GPIO 15 and make the following changes to the example code to trigger a continuous sending of CAN packets:

```diff
diff --git a/examples/peripherals/iso11898/iso11898_filter/main.c b/examples/peripherals/iso11898/iso11898_filter/main.c
index 1b3d0ad0..99e92c10 100644
--- a/examples/peripherals/iso11898/iso11898_filter/main.c
+++ b/examples/peripherals/iso11898/iso11898_filter/main.c
@@ -75,9 +75,9 @@ int main(void)
     bflb_iso11898_txint_mask(iso11898, false);
     bflb_iso11898_rxint_mask(iso11898, false);
     bflb_irq_enable(iso11898->irq_num);
-    bflb_iso11898_send(iso11898, &msg_tx, 1000);

     while (1) {
+        bflb_iso11898_send(iso11898, &msg_tx, 1000);
         bflb_mtimer_delay_ms(2000);
     }
 }
```

The oscilloscope does not register no signals:

[image]

Can you explain why that would be? Have you tested it yourself at your end after making those changes to the SDK and recompiling/flashing @sakumisu?

Btw, thanks for the making the `make flash` command work, that's good UX progress! :)

sakumisue, 2023-03-06:
[image]

sakumisue, 2023-03-06:
No problem.

sakumisu, 2023-03-06:
This has already tested before, and i think you should check by yourself, i have no time to help you now.

sakumisu, 2023-03-06:
And i see you do not use another can module, it does not work.

brainstorm, 2023-03-14:
Thanks for re-opening this issue. Even if there's no CAN transceiver attached there should be signals showing up on the oscilloscope anyway.

sakumisue, 2023-03-15:
> Thanks for re-opening this issue. Even if there's no CAN transceiver attached, there should be signals showing up on the oscilloscope anyway.

Follow our ic designer, he said.

brainstorm, 2023-03-15:
> Follow our ic designer, he said.

Excuse me, I cannot understand what you just said: what do you mean? 

I've worked successfully with other ICs that support CAN (i.e TWAI on Espressif) and the signals before and after the transceiver IC ([i.e TJA1043](https://www.nxp.com/docs/en/data-sheet/TJA1043.pdf)) can be seen on RXD/TXD as well as CAN_H/CAN_L pins, here's my scope dissecting arbitrary CAN packets on both ends on an ESP32S2:

[image]

Can you show a similar result with your ISO11898 peripheral code? Is the BL616 SDK well tested for this peripheral?

sakumisu, 2023-03-15:
Yes, but you need can transceiver. i think this issue is done.

sakumisu, 2023-03-15:
You can use any gpio pin for can tx and rx.

sakumisu, 2023-03-15:
SDK defined is GPIO14 and GPIO15

brainstorm, 2023-03-15:
> Yes, but you need can transceiver. i think this issue is done.

I've just posted a screenshot of my oscilloscope telling you otherwise, can you please test it on your end without a CAN transceiver attached? It should display the signals shown above in yellow and blue (roughly).

brainstorm, 2023-03-15:
> SDK defined is GPIO14 and GPIO15

Oh, that's news, will try that then, thx... couldn't see those on the recent code/commits.

brainstorm, 2023-03-15:
> SDK defined is GPIO14 and GPIO15

My mistake, I meant those, editing the original entry...

sakumisu, 2023-03-15:
I have said, if you try without transceiver IC, it has no signals.

brainstorm, 2023-03-15:
> I have said, if you try without transceiver IC, it has no signals.

I understood fully what you said and I'm telling you it makes no sense because the latest oscilloscope photo is telling you the opposite, what more evidence do you need?

sakumisu, 2023-03-15:
You test is other ics not ours.

brainstorm, 2023-03-15:
> You test is other ics not ours.

Fine, fair enough, I'll attach a transceiver and come back to you with more evidence, I hope I'm wrong :)

sakumisu, 2023-03-15:
And your other ic test with transceiver IC, you can remove them to test.

sakumisue, 2023-03-15:
If you use transceiver IC,it will work.
[image]

closed 2023-03-31 by sakumisue