bouffalolab/bouffalo_sdk #103: Why SDK Readme not show MacOS tool chain

PMLSTaisen, 2023-03-01:
I do find the MacOS tool chain from [https://dev.bouffalolab.com/download](https://dev.bouffalolab.com/download) , but why Readme only list tool chain for Windows and Linux? Person like me take very long time to find our there is MacOS tool chain ready. Could you add into github and make link easy for everyone?

sakumisu, 2023-03-01:
Build by yourself

sakumisu, 2023-03-01:
T-head official does not provide mac toolchain.

PMLSTaisen, 2023-03-01:
Any instruction to build under MacOS for toolchain? Under the [https://dev.bouffalolab.com/download](https://dev.bouffalolab.com/download) this webpage, why it list a MacOS toolchain? `toolchain_riscv_sifive_macos` is different from T-Head version Windows?

sakumisu, 2023-03-01:
It is different.

sakumisu, 2023-03-01:
Follow this: https://github.com/p4ddy1/pine_ox64/blob/main/build_toolchain_macos.md

brainstorm, 2023-03-04:
This issue seems to be a duplicate of my own on https://github.com/bouffalolab/bl_mcu_sdk/issues/83.

closed 2023-03-04 by PMLSTaisen