bouffalolab/bouffalo_sdk #105: [Info] Enable the LP core of BL808 and receive "Hello world" msg from LP core

cjacker, 2023-03-01:
Unfortunately, up to now, there is no official examples to show how to use LP core of BL808.

Here is a demo 'tripplecore' for it: https://github.com/cjacker/opensource-toolchain-bouffalo-lab/tree/main/triplecore_bl808

## Install XuanTie RISC-V toolchain

```
wget https://occ-oss-prod.oss-cn-hangzhou.aliyuncs.com/resource//1663142243961/Xuantie-900-gcc-elf-newlib-x86_64-V2.6.1-20220906.tar.gz
sudo mkdir -p /opt/xuantie-riscv64-embed-toolchain
sudo tar xf Xuantie-900-gcc-elf-newlib-x86_64-V2.6.1-20220906.tar.gz -C /opt/xuantie-riscv64-embed-toolchain --strip-components=1
```

And add `/opt/xuantie-riscv64-embed-toolchain/bin` to PATH env according to your shell.

## Apply patch to bl_mcu_sdk

You need apply 'bl_mcu_sdk-enable-bl808-lp-core-and-example.patch' first.

```
$ git clone https://github.com/bouffalolab/bl_mcu_sdk
$ cd bl_mcu_sdk
$ cat bl_mcu_sdk-enable-bl808-lp-core-and-example.patch | patch -p1
```

## Build this triplecore demo

I put `bl_mcu_sdk` at home dir, if not, change this line in Makefile to your sdk path:
```
make -C $@ BL_SDK_BASE=$(HOME)/bl_mcu_sdk
```

And type `make` directly.

After build successfully, these firmware files will be generated:

- ./helloworld_m0/build/build_out/helloworld_bl808_m0.bin : firmware for m0
- ./helloworld_lp/build/build_out/helloworld_bl808_lp.bin : firmware for lp
- ./helloworld_d0/build/build_out/helloworld_bl808_d0.bin : firmware for d0

## Program to bl808

Since BLDevCube GUI tool not support Group2 in BL808 MCU page, you can not 
program these firmwares in one time.

I suggest to program it one by one by `BLFlashCommand`, you may need to activate UART programming mode every time.

Here use Sipeed M1S Dock as example.

- Connect 'UART' TypeC port of M1S Dock to PC USB port.
- Hold 'BOOT' down and press 'RESET' button.
- Release 'RESET' button, then release 'BOOT' button.

M1S Dock will enter UART programming mode.

Then:
```
cd helloworld_m0
make flash BL_SDK_BASE=$(HOME)/bl_mcu_sdk
```

After programmed, you need re-power M1S Dock and continue programming firmwares for d0 and lp core as above steps.

## Wire up USB2TTL adapters

After all three firmware programmed, Wire 3 USB2TTL adapter as:

- UART0 for D0 core
  + GPIO 14 (TX) -> USB2TTL RX
  + GPIO 15 (RX) -> USB2TTL TX

- UART1 for M0 core
  + GPIO 16 (TX) -> USB2TTL RX
  + GPIO 17 (RX) -> USB2TTL TX

- UART2 for LP core
  + GPIO 18 (TX) -> USB2TTL RX
  + GPIO 19 (RX) -> USB2TTL TX

For M1S Dock, since it already have 2 UARTs connect to D0 and M0, you only need wire 
another USB2TTL adapter to GPIO 18 and GPIO 19 for LP core.

For my env, the 3 serial devices are : 

- '/dev/ttyUSB0' for D0
- '/dev/ttyUSB1' for M0
- '/dev/ttyACM0' for LP

Check it according to your env.

## Receive hello world from 3 cores

Use any serial terminal such as `tio` / `minicom` / `picocom` / `cutecom`, etc.
Set baudrate to 2000000, 8N1, and open 3 serial devices.

You will receive "hello world" from 3 cores.

[image]

## More

- GPIO 18 (TX) and GPIO 19 (RX) for LP core UART is defined in `bsp/board/bl808dk/board.c`, you can change it as you like.
- LP core is boot by M0 core, and boot from 0x58020000, FW addr set to 0x20000 in `helloworld_lp/flash_prog_cfg.ini` for `BLFlashCommand`

sakumisu, 2023-03-01:
Hi, @cjacker thank you for your effort, you do the same code as mine, could you please pr this patch into master?

cjacker, 2023-03-01:
OK, I will send a PR.

cjacker, 2023-03-01:
By the way, could you update BLDevCube for various platforms to add 'Group2' support to MCU page? After that, these firmwares can be programmed in one time.

Fishwaldo, 2023-03-01:
There still is a issue with overlapping RAM regions in the linker scripts for M0/LP

If M0 uses more than 16K ram, it will walk over the memory for LP

closed 2023-04-01 by sakumisue