bouffalolab/bouffalo_sdk #107: usbd_audio_v1_mic_speaker_multichan PC无法识别

zhuangzard, 2023-03-01:
你好，尝试烧录usbd_audio_v1_mic_speaker_multichan实例到M0Sense板子上，PC没有识别出任何的设备，这个示例还需要别的驱动吗？我用blisp flash的板子。

gamelaster, 2023-03-01:
Do you use Windows 7 ? Did you triggered BOOT mode?

zhuangzard, 2023-03-01:
I am using Win11, and yes I did triggered Boot Mode. Looks the issue comes from Clion Cmake project I created. I used other sample and blisp flash it works very well. Here what I did:

Clion Cmake file Project:

I modified the Cmakelist.txt file
```
cmake_minimum_required(VERSION 3.15)
set(CHIP bl702)
set(BOARD bl702dk)
set(CROSS_COMPILE riscv64-unknown-elf-)
set(CONFIG_ROMAPI               1)
#set(CONFIG_USB_HS               0)
#set(CONFIG_TLSF                 1)
#add_definitions()
include(proj.conf)

find_package(bouffalo_sdk REQUIRED HINTS $ENV{BL_SDK_BASE})

sdk_add_include_directories(.)

target_sources(app PRIVATE audio_v1_mic_speaker_multichan_template.c)
sdk_set_main_file(main.c)

project(usbd_audio_v1_mic_speaker_multichan)
```
Then I used Blisp to flash the chip, with Triggered Boot Mode. 
It shown flash finished. But system not responding to any USB.

If I use windows command prompt: ` make CHIP=bl702 BOARD=bl702dk` and flash it with `make flash CHIP=bl717 COMX=COM8`, it work very well.

Then I tried to flash the `bin `file generated by Clion with Cmakelist.txt with make flash command, Not work, either try flash with blisp to flash Make file generated bin file, Not work either. 

Will dig little deeper and see why this happen.

gamelaster, 2023-03-01:
@zhuangzard if you use latest MCU sdk, please try to flash it via DevCube 1.8.3/1.8.2. (or you can use blisp iot command with offset 0x0000).

zhuangzard, 2023-03-01:
I am using latest MCU sdk and use the `DevCube` under `tools ` folder. 

If I use Clion CMake to generate Bin file, not work, if I use `Makefile` directly to build the project, works fine. DevCube out of the box works all fine for flashing. 

They should work in the same way because eventually they all use Cmake file to generate first and then build with gcc, why the generated Bin file works differently?

gamelaster, 2023-03-01:
@zhuangzard The latest SDK have pre-processing tools, which adds boot header directly into the binary.
blisp is not compatible fully with this (this change was introduced few days ago), so it needs modification.

zhuangzard, 2023-03-01:
@gamelaster thank you for the reminder. However, I think something about `CMake` files under the folder `cmake` , some tool make the results different.

sakumisue, 2023-03-02:
Check your bin from clion and cmd.

sakumisue, 2023-03-02:
并且 702 使用 audio 或者video只能枚举，不能进行传输，驱动没有开放

zhuangzard, 2023-03-02:
@sakumisue 谢谢回复，正在尝试用BL702芯片做耗材设备上的mic和lcd的驱动芯片，价格和功能都很合适。看来现在还不能用audio的传输。能不能给一个相关方法去检查一下clion生成的bin，对cmd不是很了解。我把clion生成的bin复制到make 生成的文件夹里，仅仅替换bin 文件，其他的efulse没有替换，make flash烧录后主机不能对usb有反应。但是make生成的烧录没问题。

sakumisu, 2023-03-02:
对比一下bin文件内容不就知道问题了吗？

zhuangzard, 2023-03-02:
@sakumisue 大神，关键不就是不知道如何对不吗。。。

sakumisu, 2023-03-06:
> @sakumisue 大神，关键不就是不知道如何对比吗。。。

bin 文件如何对比，这个问题。。。请不要问出来

sakumisu, 2023-03-06:
对比软件都带bin文件对比功能

closed 2023-03-06 by sakumisue

sakumisu, 2023-03-06:
如果你用clion,光编译成功代码没有用，还需要对bin进行处理，我们提供的sdk里使用 make combine,你缺少的应该是这个

zhuangzard, 2023-03-06:
谢谢 @sakumisue ，应该是你说的缺少make combine。我看看能不能在clion里面加入make combine这个步骤。