bouffalolab/bouffalo_sdk #108: PT Check Failed when flashing with BLDevCube

Fishwaldo, 2023-03-02:
Hi,
Trying to flash both M0/D0 at the same time. After compiling and the bflb_fw_post_proc has run on both outputs:
```
[100%] Built target helloworld_bl808_d0.elf
[13:53:03.351] - Chipname: bl808
[13:53:03.352] - Board config dir: ./../../../bsp/board/bl808dk/config
[13:53:03.352] - [Warning] No partiton file found in ./../../../bsp/board/bl808dk/config,go on next steps
[13:53:03.352] - [Warning] No dts file found in ./../../../bsp/board/bl808dk/config,go on next steps
[13:53:03.353] - [Warning] No boot2/bootloader file found in ./../../../bsp/board/bl808dk/config,go on next steps
[13:53:03.353] - [Warning] No mfg file found in ./../../../bsp/board/bl808dk/config,go on next steps
[13:53:03.354] -
Process /home/fish/bl808/bl_mcu_sdk/examples/bl808_triplecore/helloworld_d0/build/build_out/helloworld_bl808_d0.bin
[13:53:03.355] - ========= sp image create =========
[13:53:03.355] - Flash config crc: b'b5fec518'
[13:53:03.355] - Clock config crc: b'0a894b86'
[13:53:03.355] - Image Offset:0x1000
[13:53:03.355] - Image hash is b'644ce13d777b7fd29c0d090d3dd684d75a5f4675d8e6dab8acb3ca165fd01542'
[13:53:03.356] - Encrypt efuse data
[13:53:03.357] - Image Offset:0x1000
[13:53:03.357] - Image hash ignore,not calculate
[13:53:03.358] - Bootheader config crc: b'178ecd42'
Built target combine
make[1]: Leaving directory '/home/fish/bl808/bl_mcu_sdk/examples/bl808_triplecore/helloworld_d0'
```

and: 
```
[100%] Built target helloworld_bl808_m0.elf
[13:53:00.685] - Chipname: bl808
[13:53:00.685] - Board config dir: ./../../../bsp/board/bl808dk/config
[13:53:00.685] - [Warning] No partiton file found in ./../../../bsp/board/bl808dk/config,go on next steps
[13:53:00.685] - [Warning] No dts file found in ./../../../bsp/board/bl808dk/config,go on next steps
[13:53:00.685] - [Warning] No boot2/bootloader file found in ./../../../bsp/board/bl808dk/config,go on next steps
[13:53:00.685] - [Warning] No mfg file found in ./../../../bsp/board/bl808dk/config,go on next steps
[13:53:00.685] -
Process /home/fish/bl808/bl_mcu_sdk/examples/bl808_triplecore/helloworld_m0/build/build_out/helloworld_bl808_m0.bin
[13:53:00.686] - ========= sp image create =========
[13:53:00.686] - Flash config crc: b'b5fec518'
[13:53:00.686] - Clock config crc: b'0a894b86'
[13:53:00.686] - Image Offset:0x1000
[13:53:00.686] - Image hash is b'c4d74a4df0249174e1fb55f25f3b45d5e5da16184feba9b726d105fa5254af7a'
[13:53:00.686] - Encrypt efuse data
[13:53:00.687] - Image Offset:0x1000
[13:53:00.687] - Image hash ignore,not calculate
[13:53:00.687] - Bootheader config crc: b'a940ee05'
Built target combine
make[1]: Leaving directory '/home/fish/bl808/bl_mcu_sdk/examples/bl808_triplecore/helloworld_m0'
```

I then try to flash with BLDevCube 1.8.3 with M0 in in Group0 and D0 in Group1:

```
[13:59:55.898] - EFUSE_CFG
[13:59:55.899] - BOOTHEADER_GROUP0_CFG
[13:59:55.899] - BOOTHEADER_GROUP1_CFG
[13:59:55.926] - Create bootheader using C:\Users\justi\Desktop\blcube\chips\bl808\img_create_mcu/efuse_bootheader_cfg.ini
[13:59:55.927] - Updating data according to <C:\Users\justi\Desktop\blcube\chips\bl808\img_create_mcu/efuse_bootheader_cfg.ini[BOOTHEADER_GROUP0_CFG]>
[13:59:56.065] - Created file len:352
[13:59:56.073] - Create bootheader using C:\Users\justi\Desktop\blcube\chips\bl808\img_create_mcu/efuse_bootheader_cfg.ini
[13:59:56.075] - Updating data according to <C:\Users\justi\Desktop\blcube\chips\bl808\img_create_mcu/efuse_bootheader_cfg.ini[BOOTHEADER_GROUP1_CFG]>
[13:59:56.081] - Created file len:352
[13:59:56.084] - Create efuse using C:\Users\justi\Desktop\blcube\chips\bl808\img_create_mcu/efuse_bootheader_cfg.ini
[13:59:56.087] - Updating data according to <C:\Users\justi\Desktop\blcube\chips\bl808\img_create_mcu/efuse_bootheader_cfg.ini[EFUSE_CFG]>
[13:59:56.093] - Created file len:256
[13:59:56.108] - Image create path: C:\Users\justi\Desktop\blcube\chips\bl808\img_create_mcu
[13:59:56.109] - Config file: C:\Users\justi\Desktop\blcube\chips\bl808\img_create_mcu/img_create_cfg.ini
[13:59:56.138] - img already have bootheader
[13:59:56.139] - Write flash img direct
[13:59:56.164] - img already have bootheader
[13:59:56.165] - Write flash img direct
[13:59:56.170] - ========= eflash loader config =========
[13:59:56.186] - pt check fail, chips\bl808\img_create_mcu\bootinfo_group1.bin is overlayed with chips\bl808\img_create_mcu\img_group0.bin in flash layout, please check your partition table to fix this issue
[13:59:56.187] - PT Check Fail
```

It appears that any firmware assigned to Group1 fails with the same message:
```
[14:01:45.505] - pt check fail, chips\bl808\img_create_mcu\bootinfo_group1.bin is overlayed with chips\bl808\img_create_mcu\img_group1.bin in flash layout, please check your partition table to fix this issue
[14:01:45.506] - PT Check Fail
```

Is something broken with Group1?

sakumisu, 2023-03-03:
No, devcube will start another cpu if you use two group, but in board ,it will start other cpu, so ,it is conflict， please use `make flash` in sdk.

sakumisu, 2023-03-06:
You can use make flash to download three bin.

sakumisu, 2023-03-06:
in bl808_triplecore dir, please do not use regex, and input the bin name. 
```
[FW1]
filedir = ./build/build_out/helloworld_bl808_m0.bin
address = 0x000000

[FW2]
filedir = ./build/build_out/helloworld_bl808_d0.bin
address = 0x100000

[FW3]
filedir = ./build/build_out/helloworld_bl808_lp.bin
address = 0x20000

sakumisu, 2023-03-06:
If you use regex with CHIPNAME, it will add  chipname and cpu_id suffix to bin

closed 2023-04-01 by sakumisue