bouffalolab/bouffalo_sdk #111: failed to use 'tgt_702.cfg' with RISC-V OpenOCD to debug BL702

cjacker, 2023-03-04:
I tested with XT-ZB1 devkit from aliexpress with:

- csky debug server with HLK CK-Link Lite debugger
- csky debug server with Sipeed RV Debugger Plus (with ck-link lite firmware)
- FT2232D with OpenOCD using tgt_702.cfg
- Tigard (FT2232H) with OpenOCD using tgt_702.cfg

The connection to the target device of C-Sky Debug Server or RISC-V OpenOCD is successful without any error.

**But BL702 failed to stop at any breakpoint when 'continue'.** I can only use 'n' or 's'.

The same combinations works well with BL602.

Anybody succeeded on BL702 debugging?

gamelaster, 2023-03-04:
Hello, I was debugging both BL60X and BL70X without any issues. I used always openocd, and breakpoints always worked wonderful. I used same cfg as you.

Serjios74, 2023-03-04:
> I tested with XT-ZB1 devkit from aliexpress with:
> 
> * csky debug server with HLK CK-Link Lite debugger
> * csky debug server with Sipeed RV Debugger Plus (ck-link lite firmware)
> * RV OpenOCD with Sipeed RV Debugger Plus (jtag firmware)
> * RV OpenOCD with FT2232D using tgt_702.cfg
> * RV OpenOCD with Tigard (FT2232H) using tgt_702.cfg
> 
> The connection to the target device of C-Sky Debug Server or RISC-V OpenOCD is successful without any error.
> 
> **But BL702 failed to stop at any breakpoint when 'continue'.** I can only use 'n' or 's'.
> 
> I am writing a tutorial for BL series chip, please refer to here for details how I wired up and debug : https://github.com/cjacker/opensource-toolchain-bouffalo-lab/blob/main/README.md#debugging
> 
> The same combinations works well with BL602.
> 
> Anybody succeeded on BL702 debugging?

cjacker, 2023-03-04:
> Hello, I was debugging both BL60X and BL70X without any issues. I used always openocd, and breakpoints always worked wonderful. I used same cfg as you.

Thanks for your reply.

I tested with a M0sense again, seems work.  M0sense is not very convenient, since enter UART ISP mode need connect boot to 3v3 each time, I rarely use it.

I am not sure it's a hardware issue or not,  I will close this issue now.

closed 2023-03-04 by cjacker

gamelaster, 2023-03-04:
> I tested with a M0sense again, seems work. M0sense is not very convenient, since enter UART ISP mode need connect boot to 3v3 each time, I rarely use it.

huh? What you exactly mean? Please, note that you cannot reset the board via button, since that will reset JTAG core too. You need to reset board via JTAG.