bouffalolab/bouffalo_sdk #115: bflb_fw_post_proc: Please ship sources not binaries

brainstorm, 2023-03-04:
After compiling an example successfully I now see the following errors (were not there a couple of weeks ago):

```
[100%] Built target iso11898_selftest_bl616.elf
/bin/sh: ./../../../../tools/bflb_tools/bflb_fw_post_proc/bflb_fw_post_proc: cannot execute binary file
make[4]: *** [CMakeFiles/combine] Error 126
make[3]: *** [CMakeFiles/combine.dir/all] Error 2
make[2]: *** [CMakeFiles/combine.dir/rule] Error 2
make[1]: *** [combine] Error 2
make: *** [build] Error 2
```

Then I realised that you ship binaries for 2 platforms instead of just having the source code for those tools compiled alongside?:

```
% file ../../../../tools/bflb_tools/bflb_fw_post_proc/*
../../../../tools/bflb_tools/bflb_fw_post_proc/bflb_fw_post_proc:     ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=4e2fd900053073a864e7e386b70f8ef7ae0df959, stripped
../../../../tools/bflb_tools/bflb_fw_post_proc/bflb_fw_post_proc.exe: PE32 executable (console) Intel 80386, for MS Windows

% uname -a
Darwin m1.local 22.3.0 Darwin Kernel Version 22.3.0: Mon Jan 30 20:39:35 PST 2023; root:xnu-8792.81.3~2/RELEASE_ARM64_T8103 arm64 arm Darwin
```

Could you please delete the binaries and add source code that produces those binaries instead?

Similar story goes to the CMake shipped as a binary:

```
% ls -alh ../../../../tools/cmake/bin 
total 11M
drwxr-xr-x 4 rvalls staff 128 Feb 18 21:58 .
drwxr-xr-x 4 rvalls staff 128 Jan 28 08:10 ..
lrwxr-xr-x 1 rvalls staff  23 Jan 28 13:38 cmake -> /opt/homebrew/bin/cmake
-rw-r--r-- 1 rvalls staff 11M Jan 28 08:10 cmake.exe
```

The user's system `CMake` should be used instead.

cjacker, 2023-03-04:
Agree !

The development echo-system of BL chips is messed up.

There are:
- 2 sdks : bl_iot_sdk / bl_mcu_sdk, which differ a lots.
- 4 official flash tool: BLDevCube / BLFlashCommand / bflb-mcu-tool /bflb-iot-tool.  
  + BLDevCube and BLFlashCommand are close sourced.
  + what's the difference between bflb-mcu-tool with bflb-iot-tool? 
  + no one can replace others. 
  + no one acts full functionality except BLDevCube, but it is not for cli.
- a lot of third party flash tool. OK, we can ignore them...

Expected:
- one sdk 
- one opensource flash tool with cli and gui support.

sakumisu, 2023-03-05:
- Iot sdk does not update at all, please use mcu sdk(it will be named bouffalosdk)
- please use BLFlashCommand from now, cli and gui support
- bflb-mcu-tool flashes only firmware, bflb-iot-tool flashes firmware + boot2+dts+partition+other thirty-party firmware.

sakumisu, 2023-03-05:
We do not want to depend on python environment so provide bin.

sakumisu, 2023-03-05:
Flash tool does not release with open, thanks.

gamelaster, 2023-03-05:
BLFlashCommand is also GUI?

sakumisu, 2023-03-05:
> BLFlashCommand is also GUI?

Will push gui.

sakumisu, 2023-03-05:
make with CAMKE= will use user cmake.

cjacker, 2023-03-06:
> Flash tool does not release with open, thanks.

Is BL lab sure about that? 

And did BL Lab checked and verified the LICENSEs of all relates opensource project used and packaged in BLFlashCommand and Cube?

I want to make sure that  BL lab already decided to close BLFlashCommand/Cube sources? or it will be released some day in future?

sakumisu, 2023-03-06:
I have said, if you use mcu sdk, you can use our new flash tool.

sakumisu, 2023-03-06:
Do not support multi groups, thanks. It is history.

sakumisu, 2023-03-06:
#108

cjacker, 2023-03-07:
OK, I understand, BLFlashCommand / Cube designed for bl_mcu_sdk build system and only work with current bl_mcu_sdk.

Would you mind help me to commit this path to `bflb-mcu-tool` and release a new version to pypi?  I did not find the upstream repo of `bflb-mcu-tool`. this patch did alter original args of bflb-mcu-tool, but export `--group1-firmware` and `--group1-addr` args as BLDevCube did and can be used to program low_load firmwares of classic `bl808_linux`.

```
--- bflb_mcu_tool.py.orig	2023-03-07 01:00:54.175808455 +0800
+++ bflb_mcu_tool.py	2023-03-07 00:56:35.977075903 +0800
@@ -2349,6 +2349,13 @@
         bflb_utils.printf("firmware is not existed")
         sys.exit(1)
 
+    if args.group1_firmware:
+        group1_firmware = args.group1_firmware.replace('~', expanduser("~"))
+    else:
+        group1_firmware = None
+    if not group1_firmware or not os.path.exists(group1_firmware):
+        bflb_utils.printf("group1 firmware is not existed")
+
     if args.dts:
         dts = args.dts.replace('~', expanduser("~"))
         if not os.path.exists(dts):
@@ -2524,10 +2531,13 @@
         config["public_key_cfg-group1"] = ''
         config["private_key_cfg-group1"] = ''
         config["img1_group"] = "group0"
-        config["img2_group"] = "unused"
+        if not args.group1_firmware:
+            config["img2_group"] = "unused"
+        else:
+            config["img2_group"] = "group1"
         config["img3_group"] = "unused"
-        config["img2_file"] = ""
-        config["img2_addr"] = ""
+        config["img2_file"] = group1_firmware
+        config["img2_addr"] = args.group1_addr
         config["img3_file"] = ""
         config["img3_addr"] = ""
         config["img1_file"] = firmware
@@ -2538,6 +2548,14 @@
                 config["img1_addr"] = "0x" + args.addr.replace("0x", "")
         else:
             config["img1_addr"] = "0x58000000"
+
+        if args.group1_addr:
+            if args.group1_addr == "0x2000" or args.group1_addr == "2000":
+                config["img2_addr"] = "0x58000000"
+            else:
+                config["img2_addr"] = "0x" + args.group1_addr.replace("0x", "")
+        else:
+            config["img2_addr"] = "0x58000000"
     else:
         bflb_utils.printf("Chip type is not correct")
         sys.exit(1)
@@ -2581,6 +2599,11 @@
                         default=firmware_default,
                         help="image to write")
     parser.add_argument("--addr", dest="addr", default="0x2000", help="address to write")
+    parser.add_argument("--group1-firmware",
+                        dest="group1_firmware",
+                        default=None,
+                        help="group1 image to write")
+    parser.add_argument("--group1-addr", dest="group1_addr", default="0x2000", help="address to write")
     parser.add_argument("--dts", dest="dts", help="device tree")
     parser.add_argument("--build", dest="build", action="store_true", help="build image")
     parser.add_argument("--erase", dest="erase", action="store_true", help="chip erase")
@@ -2604,6 +2627,9 @@
         bflb_utils.printf("Serial port is not found")
     bflb_utils.printf("Baudrate is " + str(args.baudrate))
     bflb_utils.printf("Firmware is " + str(args.firmware))
+    if args.group1_firmware:
+        bflb_utils.printf("Group1 Firmware is " + str(args.group1_firmware))
+        bflb_utils.printf("Group1 Addr is " + str(args.group1_addr))
     bflb_utils.printf("Device Tree is " + str(args.dts))
     bflb_utils.printf("==================================================")
     config = get_value(args)
```