bouffalolab/bouffalo_sdk #117: have some question with bflb_uart_putchar

ArpeggioP, 2023-03-06:
I have been looking  at https://bl-mcu-sdk.readthedocs.io/zh_CN/latest/api_reference/peripherals/uart.html
working with Uart I do get Uart poll example works fine

but I was tring to use bflb_uart_putchar function for sending data by uart to pc

int ch seems for data variable for Uart is the ch works with ascii? If I insert ch = 123  I thougt I will get " { " in my hyperterminal but I got " , ".   " , " in ascii is 44 could you tell me how the bflb_uart_putchar works? I have read the doc

thanks

sakumisu, 2023-03-06:
Solve by yourself

minkee555, 2023-03-06:
Well I was asking you politely
You don't have to be rude

I was just asking question

gamelaster, 2023-03-06:
When taking look at bflb_uart_putchar function, it don't do any changes to the input character, so it directly writes just to UART. Are you sure that you are sending correct data to the func?

minkee555, 2023-03-06:
@gamelaster @sakumisu  I see sorry about my word I understand.

I am realy into your works I was just having some issue with the uart function I allways read the doc fisrt before sending message

I was not sure about the ch works

minkee555, 2023-03-06:
I use putchar function ch value as a data input do I have to use it as pointer? I just use ch for example int ch =123; for test

minkee555, 2023-03-06:
So I was thinking that ch works for transfering byte data to uart and it will show up in terninal in ascii if I set it right?

sakumisu, 2023-03-06:
If printf log is ok, i think this function is ok.

sakumisu, 2023-03-06:
It is just a simple function, i don't think it has some issues, maybe your usb2ttl is bad.

ArpeggioP, 2023-03-06:
thanks for the help I will check my usb2ttl and I will try out printf log If you meen hellow world example is printf log It works perfectly

sakumisu, 2023-03-06:
If you uart poll works fine, it is ok.

sakumisu, 2023-03-25:
So, i think this is over, right?

ArpeggioP, 2023-03-26:
well I give up try to fix the problem I got terrible error after that when I connect the board to pc I wasn't able to control my keyboard and mouse lol  I will just close this issue

closed 2023-03-26 by ArpeggioP