bouffalolab/bouffalo_sdk #125: Kepping GPIO state after entering HBN0

Jatsekku, 2023-04-01:
Hi,
is it possible to keep state on GPIO (at least pull-up/pull-down instead of strong drive) after entering HBN0 mode?

Jatsekku, 2023-04-01:
I manged to keep pull-up of GPIO9 (BL702) by calling `HBN_Hw_Pu_Pd_Cfg(1);` 
just before entering HBN0 mode (using old bl_mcu_sdk v1.45). 
In my case I used GPIO9 as power source for external PIR sensor not as wake-up source. 
There is small spike when entering HBN0. It can be reduce using cap.

profound1987, 2023-04-03:
Hi Jatsekku
All the chips(BL602/BL702/BL616 etc.)can NOT keep the state on GPIO in HBN mode since the power domain of GPIO is powered off. You can try to use an external pull-up or pull-down resistor to achieve this.

Jatsekku, 2023-04-03:
Hi @profound1987, thank you for the response. 
As I said before I managed to keep pull-up of GPIO9 in HBN0 mode and that's good enough for my purpose.
Issue may be closed :)