bouffalolab/bouffalo_sdk #130: NPU Support

chiefnoah, 2023-04-18:
The BL808 has an NPU, but I'm not seeing any references to it here despite there being TensorFlowLite and TinyMaix in-repo. Am I just not seeing it, or is the NPU SDK not present.

sakumisu, 2023-04-19:
Refer to https://github.com/sipeed/M1s_BL808_SDK

holystone, 2023-05-03:
Native NPU LeakyRelu support? ...
BL808_RM_en_1.3.pdf was released ~2 weeks ago, & I can see that now LeakyRelu is 
supported natively on the NPU (page 653). But **BLAI_toolchain.exe** is not updated yet, 
so It complains that LeakyRelu not supported. Could you please update the BLAI_toolchain.exe?

Nold360, 2024-01-09:
> Refer to https://github.com/sipeed/M1s_BL808_SDK

sadly the interesting parts are binary :(
and no audio support afaik..