bouffalolab/bouffalo_sdk #131: 例程examples中的memheap示例程序，运行时会出现异常

CrazyShout, 2023-04-19:
[image]
GDB显示为内存对齐错误
我的开发板芯片为BL808，图中错误是在关闭tlsf功能产生，但是实测开启tlsf也会报错。

sakumisu, 2023-04-19:
不允许关闭tlsf

sakumisu, 2023-04-19:
请测试m0，而不是d0

sakumisu, 2023-04-19:
没有比tlsf.更好的内存管理

sakumisu, 2023-04-19:
另一个内存管理基本已经不维护了，没必要使用

sakumisu, 2023-04-19:
d0理论上都可以跑，但是我们并没有全部测试

sakumisu, 2023-04-19:
先把重心放在m0吧，其次d0本来就没多少ram，用内存管理浪费其实

sakumisu, 2023-04-20:
你可以把 memtest 屏蔽掉，我这边测试没有问题，可能这个写的有问题

sakumisu, 2023-04-20:
不过不屏蔽也能用，并没有异常

sakumisu, 2023-04-20:
[image]

CrazyShout, 2023-04-20:
> 不过不屏蔽也能用，并没有异常

好的，请问你这个测试是在m0上测试的吗

sakumisu, 2023-04-20:
上面已经给图了，m0  d0一起跑的

sakumisu, 2023-04-20:
都没问题

CrazyShout, 2023-04-20:
> 都没问题
都是tlsf算法吗，右图似乎是d0的测试结果？因为我的d0也也报的这样的，似乎卡在了循环中

sakumisu, 2023-04-20:
这个例程需要6 70k，你一共才50多k，怎么跑？

sakumisu, 2023-04-20:
你可以自己写mem test，这个是随便写的

sakumisu, 2023-04-20:
卡在循环就是因为没有mem啊

sakumisu, 2023-04-20:
这是纯软件的代码，跟硬件没有一点关系

sakumisu, 2023-04-20:
看了以下，需要大概2 3M的ram，测不了很正常