bouffalolab/bouffalo_sdk #139: BL808 M0 setting ram address to psram

baselsayeh, 2023-04-28:
Hello

I'm trying to set the ram_memory section for the BL808 M0 to pSRAM (address to 0x50000000), But it does not boot
```
    /*ram_memory  (!rx) : ORIGIN = 0x62030000, LENGTH = 160K + 64K - 20K - 4K - 40K*/
    ram_memory  (!rx) : ORIGIN = 0x50000000, LENGTH = 32K
```
Tried this with the helloworld example, but it does not boot, no serial output from the program
trying the helloworld example with the builtin boot2_isp_debug also resulted with a no output, but the isp_debug did output some info
```
Simple Malloc 5120
custom 0x0000
flash init 0
BLSP Boot2 start:Feb 13 2023,11:08:22
Group=1,CPU Count=3
ver:6.5.4
user_fw 
Imgload not found
SBI not found
Kernel not found
Rootfs not found
DTB not found
Active PT:0,Age 0
MFG not found
entry ID 0 found
pass param addr 62057c00,len 4
pass param addr 62057c04,len 236
pass param addr 62057cf0,len 92
Boot start
Media boot main
R header from 00010000
ps_mode 0,efuse hbn_check_sign 0
group[0] offset 00011000 ,core[0] offset 00000000 bootentry 58000000
group[0],core[1] halt
```

How can i run the examples but with pSRAM as the ram address?

baselsayeh, 2023-04-28:
> Hi,
> 
> app cannot be directly in pSRAM, there needs to be always a piece of code, which will be in SRAM, since when app is launched, the pSRAM is not configured. https://github.com/bouffalolab/bouffalo_sdk/blob/master/bsp/board/bl808dk/board.c#L249

Ok, can we somehow start running on OCRAM to get the pSRAM going, then continue to the pSRAM?
(I already tried running on the OCRAM (whats left of it) and it worked, but anything besides the hello world application wont fit in it & I need the whole pSRAM empty)

gamelaster, 2023-04-28:
Also, please note, by default, image is always living in SRAM, so you basically need to copy it into the pSRAM as far as I know.

baselsayeh, 2023-04-28:
> Also, please note, by default, image is always living in SRAM, so you basically need to copy it into the pSRAM as far as I know.

In my case, the image (code & ro data) resides on xip_memory (flash), but any data (.data, .bss, ... sections) resides at section ```ram_memory``` which is at the Wram (from that linker script ```ram_memory  (!rx) : ORIGIN = 0x62030000, LENGTH = 160K + 64K - 20K - 4K - 40K```)

I'm fine with the code running out of the flash, but not fine (for me) to put any data at the Wram

Is there anyway to simply use the pSRAM as the ```ram_memory``` storage section? Is there any way to initialize the pSRAM before anything else so that it can be written into?

closed 2023-05-04 by baselsayeh