bouffalolab/bouffalo_sdk #141: Last push broke components/utils/log/log.h

baselsayeh, 2023-05-08:
Hello
The latest push ([[sync] sync code](https://github.com/bouffalolab/bouffalo_sdk/commit/329f9969e9d4d6fc14f691c239733544bde48884)) broke the file log.h,
Now building any example will fail with ```fatal error: bflb_dbg.h: No such file or directory```
```
[ 97%] Building C object build_out/drivers/soc/bl808/std/CMakeFiles/std.dir/port/bl808_clock.c.o
[ 98%] Linking C static library ../../../../lib/libstd.a
[ 98%] Built target std
[ 99%] Building C object CMakeFiles/helloworld_bl808_m0.elf.dir/main.c.o
In file included from /tmp/obflr/bl_mcu_sdk/examples/helloworld/main.c:5:
/tmp/obflr/bl_mcu_sdk/components/utils/log/log.h:319:10: fatal error: bflb_dbg.h: No such file or directory
  319 | #include "bflb_dbg.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/helloworld_bl808_m0.elf.dir/build.make:76: CMakeFiles/helloworld_bl808_m0.elf.dir/main.c.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:513: CMakeFiles/helloworld_bl808_m0.elf.dir/all] Error 2
make[1]: *** [Makefile:91: all] Error 2
make: *** [../../project.build:73: build] Error 2

```

sakumisue, 2023-05-09:
github ci has passed.

sakumisue, 2023-05-09:
checkout code again.