bouffalolab/bouffalo_sdk #146: 请教分区表toml文件疑问

nickfox-taterli, 2023-05-18:
针对BL808

```
[[pt_entry]]
type = 16 # 这里什么意思,0是M0,1是LP?,2是D0?,8是DTS?还有其他保留定义吗?
name = "Boot2" # 这个是不是随便写的?
device = 0 # 这个从未找到解释?
address0 = 0  # 这个从未找到解释?
size0 = 0xE000 # 允许的img大小? 
address1 = 0  # 这个从未找到解释?
size1 = 0  # 这个和size0有什么关系?
# compressed image must set len,normal image can left it to 0
len = 0
# If header is 1, it will add the header.
header = 1 # header是什么也没解释
```

strongwong, 2023-05-20:
@nickfox-taterli 你好，感谢你的关注，我这里有一点信息，跟你分享一下。
type：一般是用来指示的镜像类型。0一般表示 cpu0 的启动镜像，1一般表示 cpu1 的启动镜像。boot2 会根据这个 type 来启动镜像。因此，在自定义分区表时候，要避开0和1即可，否则会被Boot2当成可运行的镜像启动运行。
name：分区名称可以自定义，应用固件一般按照分区名称来查找分区表。分区名称最大不超过8字节。
device：暂且是预留功能。
address0和address1：表示分区的起始地址；对于需要 ota 的分区，address0和address1都要填写正确的值。
size0和size1：即分别表示从address0和address1起始的分区大小。两个分区大小可以不一致。因此，对于不需要ota的分区，address1和size1都填0即可。
header：是表示该分区是否需要添加 bootheader，对于可运行的分区才需要设置为1。

nickfox-taterli, 2023-05-20:
> @nickfox-taterli 你好，感谢你的关注，我这里有一点信息，跟你分享一下。 type：一般是用来指示的镜像类型。0一般表示 cpu0 的启动镜像，1一般表示 cpu1 的启动镜像。boot2 会根据这个 type 来启动镜像。因此，在自定义分区表时候，要避开0和1即可，否则会被Boot2当成可运行的镜像启动运行。 [...]

事实上type并不是这样的,我看sipeed的库,其中2就是d0启动,而bl808有3个cpu,同时boot2这个烧录的固定内容,所以16应该也有特殊定义,而8是bl808 linux时候用到dts,我修改type后就找不到dts了,应该也有定义(或者至少在opensbi中约定了?)

明白的,那官方repo是否考虑添加3核心启动的toml参考?基于官方4M或者Sipeed 16M Flash的配置?

谢谢你的回复.

closed 2023-08-28 by sakumisue

staunchheart, 2024-05-28:
SDK中BL616 config/partition_cfg_4M.toml中
FW的Address1/sizeo与mfg的address0/size0 重叠
这是什么意思，这样不会将mfg分区覆盖吗？